hku-mars/loam_livox #5: Poor quality not matching demo material

EdisonBing, 2019-09-03:
Thank you for your excellent work but I'm unable to match results of demo material . Testing inside of an interior space 5x5x5 meters and the result looks more like a ball of yarn , Is this a limitation of Loam or something I'm doing wrong

ziv-lin, 2019-09-03:
Are you trying to improve the speed of our algorithms, by modifying the size of the cube? Dont's worry, the next version of codes will optimize the performance of our algorithms, which will be released in this month.

EdisonBing, 2019-09-03:
I was hoping to modify for use in a smaller space on a tripod mount  , speed isn’t really an issue since it will capture in increments. Excited for the next release and much appreciate your input

closed 2019-09-06 by ziv-lin