hku-mars/loam_livox #8: Georeferenced 3D Pointcloud (i.e. las/laz file) -Question not issue

ns-1m, 2019-09-26:
Hi,

I know that you are not using IMU/GPS sensor for this loam_livox, can the result of loam_livox converted in las/laz file for RS/GIS analysis? Can the loam_livox rosbag be georeferenced 3D Pointcloud?

However, if you have IMU/GPS sensor attached to livox mid-40, what would be a possible scenario regarding using this loam_lidar for georeferencing 3D pointcloud?

Thanks.

Noli

ziv-lin, 2019-09-27:
Sorry, I am not familiar with the topic about "las/laz file for RS/GIS analysis", but I can provide you some tools to save and reload the point cloud in the coming update, I think this might be useful for you. Please wait with patience~

ns-1m, 2019-09-27:
Thanks for replying.

PDAL is the tool that use for LiDAR processing.
https://pdal.io/about.html

Georeferencing
https://pdal.io/workshop/exercises/georeferencing/index.html

What I read that cartographer-ros can create georeferenced LiDAR. 
https://google-cartographer-ros.readthedocs.io/en/latest/index.html

Out of the topic. But I might be good if cheap RTK GNSS can be incorporated in Livox for georefencing Livox Lidar Pointcloud, etc.
https://community.emlid.com/t/reach-m-ros-integration/14028/5

Thanks. Noli

ziv-lin, 2019-09-27:
Thanks a lot for your materials. 
It needs some time for me to read such materials carefully, and I will consider adding such features (or providing some tools for conversion or evaluation) in the future.

ns-1m, 2019-09-27:
OK.

I just like to add this reference
Georeferenced LiDAR 3D Vine Plantation Map Generation
https://www.ncbi.nlm.nih.gov/pmc/articles/PMC3231455/

Thanks.

Huang88888888Fei, 2019-12-16:
> Sorry, I am not familiar with the topic about "las/laz file for RS/GIS analysis", but I can provide you some tools to save and reload the point cloud in the coming update, I think this might be useful for you. Please wait with patience~

Hi, I wonder how can I export the result to *.pcd files. Now I have recorded a rosbag file, and it looks good in rviz, but I can't get the coordinates of the points. How can I export them to *.pcd files? Then I can process them in pcl. Thank you!

Egor-Morozov, 2020-04-16:
Hello!
You can install https://github.com/ANYbotics/point_cloud_io 
This node can write every message of topic "/velodyne_cloud_registered" as ply cloud.
You have to edit rosbag.launch file (see example by link below)
After finish of:
===
roslaunch loam_livox rosbag.launch
rosbag play CYT_02.bag
===
you can open all ply (about 12500 for CYT_02)  by cloudcompare and merge all ply parts.
And save result as las. 
The link for example of rosbag.launch and the result - CYT_02.las = https://drive.google.com/open?id=1RhBJK3zEQTQWF9ixenswgs5YlTzAn_6p 

wbr, Egor.

AliveForU, 2020-11-23:
> > 抱歉，我不熟悉“用于RS / GIS分析的las / laz文件”主题，但是我可以在即将到来的更新中为您提供一些工具来保存和重新加载点云，我认为这可能对您有用。请耐心等待〜
> 
> 嗨，我想知道如何将结果导出到* .pcd文件。现在，我记录了一个rosbag文件，它在rviz中看起来不错，但是我无法获取这些点的坐标。如何将它们导出到* .pcd文件？然后，我可以在pcl中处理它们。谢谢！

Hi,I have the same question with you,do you have solutions now?

fjc378209410, 2021-01-14:
Hi,
I am trying to combine livox and cartographer for slam, but my livox_frame keeps beating, how did you solve this problem?