hku-mars/loam_livox #10: Mid-100 support

msadowski, 2019-10-01:
Hi!

Thanks a lot for this package. I've run some tests already with Mid-40 and really like it so far! Would you be able to give me some pointers on how to implement Mid-100 support?

If you take pull request I'd be more than happy to expose this feature as a parameter and make a PR for it.

ziv-lin, 2019-10-05:
Hi,
I have received several emails about the adoption of Livox-Mid100, and our algorithm currently not support MID100.
However, I will treat it as the first priority after I have finished the coming update (which mentioned on our project page).

ziv-lin, 2019-10-05:
To make our algorithm support for MID-100, you need to address the following issues:
1. Get the data from MID-100, and know that which they are measured by which unit. 
2. Known (or calibrate) the extrinsic between three-unit of lidar of MID-100.
3. Make some adaptation on the modules of "feature extraction" and “laser mapping”.

musicshmily, 2019-11-26:
hello,Can you give me some instructions about how I will only use one of mid-100 for mapping

musicshmily, 2019-11-26:
> Hi!
> 
> Thanks a lot for this package. I've run some tests already with Mid-40 and really like it so far! Would you be able to give me some pointers on how to implement Mid-100 support?
> 
> If you take pull request I'd be more than happy to expose this feature as a parameter and make a PR for it.

Have you tried to use one of mid-100（also mid-40） for map reconstruction

msadowski, 2019-11-26:
@musicshmily nope, sorry

ziv-lin, 2019-11-26:
I am ongoing the work of adopting our algorithm to Mid-100, and now I am sampling the data using this LiDAR. A relate solution will be released recently (**roughly in 10 days**), thank you for your waiting~.
[image]

musicshmily, 2019-11-26:
Thank you very much and look forward to your outstanding work

musicshmily, 2019-11-26:
> @musicshmily nope, sorry

That's all right, thank you

ziv-lin, 2019-12-06:
I have made my algorithm capable work with Mid-100 and this version will be released recently.
One of the results is shown as following:
[image]

ziv-lin, 2019-12-21:
Our package support Livox Mid-100 now, please update your version the newest one and kindly refer to the README.md and Tutorial_Mid100.md to launch our package with your Livox-Mid100

musicshmily, 2019-12-21:
thank you very much,i 'll try this latter

closed 2019-12-29 by ziv-lin