hku-mars/loam_livox #11: Interfacing to ROS nav stack (Question)

griz1112, 2019-10-04:
Trying to start up ROS navigation stack with output from loam-livox. Not having much luck. Is there a document that explains the topics published by loam-livox? Can you point me to resources explaining how to do this?

ziv-lin, 2019-10-06:
Hi~,
Sorry for replying your message/issue/question so late.
For getting the point cloud, you can subscribe to the rostopic "/velodyne_cloud_registered".
For odometry, you can subscribe to the rostopic "/aft_mapped_to_init".

griz1112, 2019-10-07:
I have it working now. Using velodyne_cloud_registered as point cloud input to octomap_server its producing 2-d occupancy maps now. Adding in pointcloud_to_laserscan for a flat laser scan amcl has all it needs to kick off the nav stack.

closed 2019-10-07 by griz1112