hku-mars/loam_livox #12: What camera did you choose ?

chenkaiC4, 2019-10-05:
Hi, NICE WORK! but what camera did you choose to fit the 3D printing toos?

ziv-lin, 2019-10-05:
Sorry, my mistake. I will add a readme (project description) for that repo today, which will list all of the materials and devices.

ziv-lin, 2019-10-05:
Hi~
The project description of our handheld device is updated just now, you can now figure our which camera we used from "material lists".
[On GitHub of my personal project](https://github.com/ziv-lin/My_solidworks/tree/master/livox_handhold).
[On GitHub of our laboratory ](https://github.com/hku-mars/Livox_handheld).

ns-1m, 2019-10-05:
What is the specs of this usb camera. The TaoBao link is in chinese.

Can this one be used. https://www.aliexpress.com/item/32863177494.html?spm=a2g0o.productlist.0.0.5eb93a1e9Q0DiX&algo_pvid=8dc6e6b0-f7c5-4a87-8055-97ddb6a71a6f&algo_expid=8dc6e6b0-f7c5-4a87-8055-97ddb6a71a6f-17&btsid=bf13ccc7-72e4-4f11-9ae3-1b570b5339ca&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_53

ns-1m, 2019-10-05:
How about this one?

https://www.aliexpress.com/item/32340358933.html?spm=a2g0o.productlist.0.0.29f830c6WNKnll&algo_pvid=92781e00-1eea-4094-8a89-91258e83424c&algo_expid=92781e00-1eea-4094-8a89-91258e83424c-13&btsid=fffb73fb-9694-45b0-9b5f-a6a3e32d331e&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_53

ziv-lin, 2019-10-06:
Hi,
Actually, I have no idea to determine if these two cameras are OK or not. since it is hard for me to tell if their size is suitable for our printable modules. 
However, if you have Solidworks installed on your computer, you can tune the size of our module (**camera_holder.sldprt**) to make it suitable for your camera.
Otherwise, you can temporarily install your camera by sticking it using double-sided tape (or 3M tape like this [one](https://www.aliexpress.com/item/32418742004.html?spm=a2g0o.productlist.0.0.1bc743f3YbAIDQ&algo_pvid=f4b20b5e-9918-48f3-b13c-4adf650d8e91&algo_expid=f4b20b5e-9918-48f3-b13c-4adf650d8e91-0&btsid=0834af61-e792-48a4-921b-fe6fb9436fd0&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_52)  )

chenkaiC4, 2019-10-08:
@ziv-lin thank you!

closed 2019-10-14 by chenkaiC4