hku-mars/loam_livox #14: bag path do not exist

Niexiaoer, 2019-10-05:
I try to roslaunch the program with the downloaded file and try to input like this:
`roslaunch loam_livox rosbag.launch rosbag play /home/Downloads/CYT_02.bag`
get the roslaunch error:  The following input files do not exist: /home/Downloads/CYT_02.bag, I don't know why?

ziv-lin, 2019-10-05:
Hi~
The bag file "CYT_02.bag" is not in the directory "/home/Downloads/" (and I guess the right path is "/home/XXX/Downloads/", where XXX is your user name), please give the path correctly.

Niexiaoer, 2019-10-05:
Thank you for your help, I have found my fault. It’s not about the path of the bag file, I didn’t know that I should open another terminal to do the second line command. Now it works.

ziv-lin, 2019-10-06:
OK, glad to hear that you have resolved your problem:)

closed 2019-10-06 by ziv-lin