hku-mars/loam_livox #16: time Map corner and surf num are not enough?

OceanWong1991, 2019-10-22:
Hi, 
 when I run "roslaunch loam_livox livox.launch
"
the terminal print 

`[ WARN] [1571727835.757896886]: time Map corner and surf num are not enough
****** min max timestamp = [29.464680, 29.512922] ****** 
map corner num 0  surf num 108 
[ WARN] [1571727835.807914851]: time Map corner and surf num are not enough
****** min max timestamp = [29.512922, 29.558619] ****** 
map corner num 0  surf num 108 
[ WARN] [1571727835.858688485]: time Map corner and surf num are not enough
****** min max timestamp = [29.558619, 29.609156] ****** 
map corner num 0  surf num 108 
[ WARN] [1571727835.908180349]: time Map corner and surf num are not enough
^C****** min max timestamp = [29.609156, 29.669964] ****** 
map corner num 0  surf num 108 
[ WARN] [1571727835.958653270]: time Map corner and surf num are not enough
`

Is that something wrong?

ziv-lin, 2019-10-22:
The appearance of this warning indicates that the feature (edge features and planar feature) in your sampling scenarios are not sufficient, can you change the place of data sampling? We strongly recommend you to use this LiDAR on outdoor to avoid the case of degeneration.

OceanWong1991, 2019-10-23:
@ziv-lin Thx for your replay! I will try it

closed 2019-10-23 by OceanWong1991