hku-mars/loam_livox #17: Run KITTI Dataset

FatEvilCat, 2019-10-22:
Thanks for sharing code.
I tried to use this code to run KITTI Data. I changed three places of rosbag.launch:
1) scan_line -> 64
2) lidar_type -> velodyne
3) <remap from="/laser_points" to = "/livox/lidar" />  ->  <remap from="/laser_points" to = "/kitti/velo/pointcloud" />
but nothing shows up. I don't know why. Please help me.

ziv-lin, 2019-10-22:
KITTI datasets are sampled with Velodyne LiDARs, which our algorithm currently not support. 
However, we will adopt these LiDAR in our future work.

closed 2019-10-28 by ziv-lin

eros-ck, 2020-05-13:
This is really a lazy question @FatEvilCat . Before asking for a free lunch, please look at the code and try to understand how it actually works instead of taking someone else work without any efforts to work for you :)

eros-ck, 2020-05-13:
BTW, thanks @ziv-lin for your impressive work and sharing with the community!

gzchenjiajun, 2022-01-26:
Is there a suitable solution for livox dataset Ccnvert kitti dataset now?

bruce-almon, 2022-01-26:
> Is there a suitable solution that convert kitti dataset to livox dataset now?

I am trying to convert from livox dataset to kitti...

gzchenjiajun, 2022-01-26:
@bruce-almon 
I am also doing this work recently,may I have a reference to your code?

gzchenjiajun, 2022-02-21:
@bruce-almon I'm working on it ..