hku-mars/loam_livox #19: Problems getting this working with ROS navigation stack

griz1112, 2019-11-14:
I'm trying to integrate loam-livox with the ROS nav stack on my rover. The stack is working with a ZED camera and Rtabmap now. When I try and switch it over to loam-livox I can see the map if I run velodyne_cloud_registered into octomap_server. So that appears to be working. However when the robot moves I see the aft_mapped odometry moving but its not moving the robot model in rviz like its supposed too. Noticed that loam-livox doesn't publish odom or map like the other mapping software I've used does. So thinking that incorrect setup of the tf is what is causing my issue. Have you used it with ROS navigation?

ziv-lin, 2019-11-17:
Sorry, I can not fully understand your problems, can you provide me more information about your issue.

griz1112, 2019-11-21:
I managed to get it working.

closed 2019-11-21 by griz1112