hku-mars/loam_livox #20: Problems building V2

griz1112, 2019-11-16:
How did you build this under ROS? When I try it fails looking for a function that isn't in the opencv libs that ROS installs. I built ROS from source and got it to compile but no livox/scan or velodyne_cloud_registered messages when I start it that way. I built ROS with opencv3.4.3 and that takes care of the missing function which was introduced in 3.3.2 from what I've read. Nvidia Xavier installs 3.3.1 as part of the flash.  If any 3.3.1 or 3.2 opencv libs exist it fails. The error is undefined reference to cv::Mat::updateContinuityFlag().

griz1112, 2019-11-16:
Managed to build the software by adding vision_opencv and image_pipeline to the livox workspace. It found and linked too the correct libraries when I did that.

closed 2019-11-16 by griz1112

ziv-lin, 2019-11-17:
Thanks for your issue/remind, I have now updated the readme.