hku-mars/loam_livox #21: Memory Leak?

griz1112, 2019-11-18:
If you leave loam-livox running for about an hour it will eventually take up all the memory and swap and crash.  This is on a nvidia Xavier which has 16Gb of ram. Ubuntu 18.04. Easy to see with jtop. It takes a long enough period of time to crash that it isn't really going to effect my testing that much but I thought you might like to know.

ziv-lin, 2019-11-19:
The memory of 16Gb is too small for ** one-hour running**...
Can you install more memory on your platform, otherwise, you can turn off the "loop closure" for memory saving.

griz1112, 2019-11-21:
Its a jetson Xavier so the memory is fixed. No biggie my rover won't run an hour so won't impact me. Just saw that it did crash eventually so I thought you might not know.

closed 2019-11-21 by griz1112