hku-mars/loam_livox #22: Possible converted bag files can output format that can be opened in CloudCompare (Not issue)

ns-1m, 2019-11-19:
Hi,

It seems that CloudCompare can open and view various XYZ data files.

https://en.wikipedia.org/wiki/CloudCompare

I think it would good if possible that converted bag files can output format that can be opened in CloudCompare. 

Thanks.

Noli

ziv-lin, 2019-11-20:
Is files with "*.ply" suffix ok? If so, I can provide some tools for saving point cloud as ply files.

ns-1m, 2019-11-20:
YES, *.ply file would be good.

Thank you very much.

Noli

vedran-damjanov, 2019-12-12:
> Is files with "*.ply" suffix ok? If so, I can provide some tools for saving point cloud as ply files.

I've managed to run your LOAM on Raspberry Pi 4 with Ubuntu 18.04. It works great !
To get an output, I've changed config file if_save_to_pcd_files: 0 to 1, but this doesn't give me end result, it writes .pcd files one by one as they get recorded. Could you please provide tool for .ply export at the end of scanning ?

Vedran

ziv-lin, 2019-12-12:
Sorry for replying this issue so late, you can use the tool provided by PCL to convert them as ply files:)
For example, you can type the commands to convert them as ply files:
``pcl_pcd2ply  input.pcd output.ply``
[image]

In addition, you can also use tools pcl_viewer (also provided by PCL) to visualize the pcd files:
like this:
``pcl_viewer input.pcd``
[image]

ziv-lin, 2019-12-12:
Also, you can directly modify to code of saving *.pcd to change it as *.ply files.

amazingye, 2020-01-06:
> Sorry for replying this issue so late, you can use the tool provided by PCL to convert them as ply files:)
> For example, you can type the commands to convert them as ply files:
> `pcl_pcd2ply input.pcd [...]

hi，thanks for your great work.
in your reply you show us how to convert a single .pcd file to .pcl , but I'm wondering how can we 
get an output .pcd file after an entire scanning work, changing config 'if_save_to_pcd_files: 0' to 1 didn't help because it writes .pcd files one by one as they get recorded, and we get tons of .pcd files instead of  one as we expected.
Any reply will be appreciated~