hku-mars/loam_livox #25: How do you recreate your work

musicshmily, 2019-11-26:
Hello, thank you for your amazing work, but I am not quite clear about what to do after connecting lidar, could you please provide a detailed operation instruction

ziv-lin, 2019-11-26:
You can directly launch our algorithm by the following step:
Connect to your PC to Livox LiDAR (mid-40) by following  [Livox-ros-driver installation](https://github.com/Livox-SDK/livox_ros_driver), then
```
    roslaunch loam_livox livox.launch
    roslaunch livox_ros_driver livox_lidar.launch
```
To improve the whole wise performance, we strongly recommend you to fresh your livox lidar with this [firmware](https://github.com/Livox-SDK/Special-Firmwares-for-Livox-LiDARs/tree/master/Threadlike-Noise_Filtering_Firmware_For_Livox_MID) by following this [tutorial](https://github.com/Livox-SDK/Special-Firmwares-for-Livox-LiDARs).

musicshmily, 2019-11-26:
Can you tell me  which firmware  should i use

ziv-lin, 2019-11-27:
Any firmware is ok with our algorithm , and to further improve the precision, decrease the noise point cloud in the maps, make our result look cleaner, we recommend you to use the  [Threadlike-Noise Filtering Firmware](https://github.com/Livox-SDK/Special-Firmwares-for-Livox-LiDARs/tree/master/Threadlike-Noise_Filtering_Firmware_For_Livox_MID).

musicshmily, 2019-11-27:
thank you,i 'll have a try

closed 2019-12-06 by ziv-lin