hku-mars/loam_livox #26: Whether can make only  one of the mid-100 work

musicshmily, 2019-11-26:
How  change your code so that can make only  one of the mid-100 work( Mid-100 is made up of three mid-40s )

ziv-lin, 2019-11-26:
Yes, Mid-100 is made up of three independent units of Mid-40 and you can launch our algorithm by using one of them.
Plus, I am ongoing the work of adopting our algorithm to Mid-100, and now I am sampling the data using this LiDAR. A relate solution will be released recently (**roughly in 10 days**), thank you for your waiting~.
[image]

musicshmily, 2019-11-26:
but when i carry out in my mid-100 by the way you said(   
 roslaunch loam_livox livox.launch
 roslaunch livox_ros_driver livox_lidar.launch )
,it does't work,phenomenon as follow:

1. **first:**
[image]

it's result:
[image]

2.**second**
[image]

it's result:
[image]

the first result is disappear,and after execute the two row code( roslaunch loam_livox livox.launch
 roslaunch livox_ros_driver livox_lidar.launch ),the result just like open the lidar drivers

Did i do something wrong?

musicshmily, 2019-11-26:
the second row code,for just make one of mid-100 work,so i use the code as follows to replace:
roslaunch livox_ros_driver livox_lidar_rviz.launch bd_list:="1LVDG1S00660271"

musicshmily, 2019-11-26:
the command windows show as follows:
[image]

ziv-lin, 2019-11-26:
It seems that there has some errors in Ceres-Solver, have you install **SuiteSparse** and **CXSparse** on your computer (these two libs are required for building ceres-solver).
I suggest you reinstall your Ceres-Solver (full version) by following instructions here:
http://ceres-solver.org/installation.html
If the issue still exists, there might be some other reason cause the problem.

musicshmily, 2019-11-26:
> It seems that there has some errors in Ceres-Solver, have you install **SuiteSparse** and **CXSparse** on your computer (these two libs are required for building ceres-solver).
> I suggest you [...]

Thank you for your advice, I will try as you said tomorrow

ziv-lin, 2019-12-06:
I have made my algorithm capable work with Mid-100 and this version will be released recently.
One of the results is shown as following:
[image]

ziv-lin, 2019-12-06:
I believe this topic is related to issue #10 , to avoid discussion the same thing among different topics, I close this issue anyway.

closed 2019-12-06 by ziv-lin

musicshmily, 2019-12-06:
Wow ,Exciting news! I am keeping an eye on your outstanding work.