hku-mars/loam_livox #27: Which version of PCL are we using in this repo?

wyw0424, 2019-11-27:
Hello,

I am using Ubuntu 16.04 LTS and installed kinetic version of ROS (which has PCL 1.7 installed). I encountered these error when I catkin_make under my catkin workspace:

/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libpcl_features.so: undefined reference to `_ZNK3pcl6search6SearchINS_11PointXYZRGBEE12radiusSearchERKNS_10PointCloudIS2_EEidRSp6vectorIiSaIiEERS8_IfSaIfEEj'
collect2: error: ld returned 1 exit status
loam_livox/CMakeFiles/livox_scanRegistration.dir/build.make:462: recipe for target '/home/derek/workspace/ros/devel/lib/loam_livox/livox_scanRegistration' failed
make[2]: *** [/home/derek/workspace/ros/devel/lib/loam_livox/livox_scanRegistration] Error 1
CMakeFiles/Makefile2:7030: recipe for target 'loam_livox/CMakeFiles/livox_scanRegistration.dir/all' failed
make[1]: *** [loam_livox/CMakeFiles/livox_scanRegistration.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

Thanks in advance!

ziv-lin, 2019-11-27:
Hi~,
We have tested our algorithm on both PCL 1.9 and PCL 1.7. 
However, I find out that the point cloud output form the voxelgrid filter vary form PCL 1.7 and 1.9, and PCL 1.7 leads some failure in some of our examples. 
In a words, if your use PCL 1.7, I strongly recommend you update your PCL as 1.9.

wyw0424, 2019-11-27:
> Hi~,
> We have tested our algorithm on both PCL 1.9 and PCL 1.7.
> However, I find out that the point cloud output form the voxelgrid filter vary form PCL 1.7 and 1.9, and PCL 1.7 leads some failure in [...]

Thank you for this quick reply! I'll get it a shot.

closed 2019-12-06 by ziv-lin