hku-mars/loam_livox #28: unable to reproduce experimental results

wlt027, 2019-11-27:
Hi,Thanks for your open source.I test the new code on Ubuntu16.04 && ROS Kinetic.But I cannot reproduce experimental results using the example data——loop-simple.bag.
Here are some screen shots.
a)loop close error
[image]

b)big delay
[image]

c)ceres-solver optimization error 
[image]

Any suggestions?

ziv-lin, 2019-11-27:
What kind of PCL version did you use?
Recently, I find out that the point cloud output form the voxelgrid filter vary form PCL 1.7 and 1.9, and PCL 1.7 leads some failure in some of our examples.
In a words, if your use PCL 1.7, I strongly recommend you update your PCL as 1.9.

wlt027, 2019-11-28:
After update my PCL as 1.9.1,problem solves.Thanks.

closed 2019-11-28 by wlt027

ziv-lin, 2019-11-28:
Thanks for your issue, I have to update the readme now.