hku-mars/loam_livox #29: Is there a data interface for velodyne lidar?

mengqingyu123, 2019-11-27:
First of all, thank you very much for the exciting work of open source. When I read the source code, I found that the Velodyne option is provided in laser_feature_extractor.hpp. I tried to modify and compile, but rviz showed nothing when playing the rosbag of the Velodyne lidar.
Is it possible to use Velodyne radar by changing the parameters? Thank you very much!

ziv-lin, 2019-11-27:
Our current version does not support for Velodyne LiDAR...
However, I have developed a temporary version (unstable, on developing) that supports velodyne  LiDAR, some  of the result (run on Kitti dataset) are shown as follow:
[image]
[image]
[image]
This version is under developing and **I will opensoure it in the coming future**.

mengqingyu123, 2019-11-27:
I look forward to your work, because our laboratory does not have a livox lidar, so I can only test your current version in the dataset.
Take the liberty to ask, is there any feasible solution to convert offline saved Velodyne lidar data into the form of livox? Still, need to modify the data interface in the program?
Thank you for your patience.

ziv-lin, 2019-11-27:
If you have a Velodyne LiDAR, you can try other open source projects, such as:
A-LOAM: https://github.com/HKUST-Aerial-Robotics/A-LOAM
Loam_velodyne: https://github.com/laboshinl/loam_velodyne
Plus, once I have finished updating the version I mentioned, I will inform you at that time :)

mengqingyu123, 2019-11-28:
Thank you for your patience, I am very interested in the method of loop detection using lidar point cloud in your work, which is not available in other algorithms.
I will keep an eye on your open source code and finish my work step by step, thank you again!

chenxianbo, 2019-12-12:
Hi, I am also very interested in loopclosure  of you work, but my sensor is velodyne,can you send the temporary version with velodyne, I want merge it to Lego_loam.

Sun1307010608, 2019-12-23:
@ziv-lin Hello, Do you compare your development version on velodyne lidar to the kitti dataset, how its localization result in the kitti ranking.

gzchenjiajun, 2022-01-26:
I also have the requirement to use kitti dataset, may I ask how this problem has been solved?