hku-mars/loam_livox #30: Drop lidar frame in mapping

Mr-Xavier, 2019-11-29:
Hi, 
thanks for your project. when I run "roslaunch loam_livox livox.launch", I get the warning like this:
**[ WARN] [1575016816.066039723]: Drop lidar frame in mapping for real time performance !!!**
Is that something wrong? Can you give me some advices?

Mr-Xavier, 2019-11-29:
The  effect of mapping is not ideal .
[image]

ziv-lin, 2019-12-06:
Sorry for replying to your issue so late since I am busying in working with my homework and exams recently.
Form the preview show in rviz, I believe you are sampling the data with objects nearby you, and the feature (edge and plane feature) it too less to solve the relative pose well.
Due to the saturation of receiver, mid40 has a bad measurement of sampling objects located nearby the LiDAR (like distant smaller than 5 meters), I suggest you test our algorithm in outdoor scenarios.

closed 2019-12-10 by ziv-lin