hku-mars/loam_livox #31: laser mapping crash after loop closure

wlt027, 2019-12-03:
Hi,I found that laser mapping will crash after loop closure.It happens both on rosbag_loop_simple and rosbag_loop.Here are some screen shots.
loop_simple
[image]
loop
[image]

Any suggestions?

ziv-lin, 2019-12-06:
Sorry for replying to your issue so late since I am busying in working with my homework and exams recently.
Yes, this actually a bug here, which relate to some operation conflicts among different thread.
I will try to figure out this bug and fix it in future updates.

wlt027, 2019-12-07:
Thank you for your repley.I'm looking forward to the next updates.

ziv-lin, 2019-12-12:
I have found out the problem, it is the deconstruction of the variable "scene_align" cause the crash. 
I will try to fix this problem in our future update...

wlt027, 2020-01-13:
problem solved

closed 2020-01-13 by wlt027