hku-mars/loam_livox #32: Running large_scale demo died.

YangSiri, 2019-12-09:
Hi, great contributors,

Today i'm testing your awesome project. I downloaded the large-scale _HKUST_01.bag_ . But it somehow corrupted after a few seconds.

This is my terminal and problem:
[image]
[image]

Looking forward to your reply.
THANKS

whiteineyes, 2019-12-10:
me the same, not just in `largescale.launch` ,but all the launches.

YangSiri, 2019-12-10:
Hi, @whiteineyes 

Did you try other ros packages like loam_velodyne?

whiteineyes, 2019-12-10:
@YangSiri nope,`cause I got one livox mid40 in hand....

whiteineyes, 2019-12-10:
@YangSiri 你是说用livox的rosbag去试验loam_velodyne?还是只是单纯的问有没有试过其他loam？

YangSiri, 2019-12-10:
Hi, @whiteineyes 

yes, the second option. i mean other method.

whiteineyes, 2019-12-10:
@YangSiri  yeah,i did test Lego_loam

whiteineyes, 2019-12-10:
@YangSiri  which PCL version r u using? 1.9.1?

YangSiri, 2019-12-10:
Hi, @whiteineyes 

1.8.1, i suppose. But isnt there a 1.7 come with ros?

whiteineyes, 2019-12-10:
1.7 come with Ros-Kinetic, 1.8 with Melodic.
I tried livox_loam  with PCL1.7 in Kinetic and 1.8 in Melodic, it went well but without loop closure;
and livox_loam version1 with pcl1.9 in Melodic,it went well too.

YangSiri, 2019-12-10:
Yes, @whiteineyes 

My env is Ros-Kinetic. I build loam_livox without any errors but i cant correctly run it with any recorded bags. It seems that _laser_Mapping_ is broken somewhere.

whiteineyes, 2019-12-10:
In  specifically, 
#### rosbag*.launch:

 1. only loop section failure in PCL1.7 and PCL1.8;

 2. the same corruption after a few seconds in PCL1.9

YangSiri, 2019-12-10:
Yes, @whiteineyes 

But i cant run it from the beginning, let alone until loop. 

It stopped as long as this :

[image]

ziv-lin, 2019-12-10:
How do you launch? for this bag, the correct configuration is:
roslaunch loam_livox rosbag_largescale.launch
rosbag play YOUR_DOWNLOADED.bag

YangSiri, 2019-12-10:
Yes @ziv-lin 

That’s exactly how I launch. As you can see from the images above.

ziv-lin, 2019-12-10:
I have no I idea why this happens, I have tested this bag file before pushed my code... 
Are you running with the new clone codes? is that other examples run successfully?

YangSiri, 2019-12-10:
Yes @ziv-lin 

I cloned the newest codes. All of examples have similar issues.

ziv-lin, 2019-12-10:
It seem that you have some issue with your environments...
1. Are you build ceres-solver with sparse(Cxsparse, suitsparse....) libraries enable?
2. Are you sure you are using the PCL with 1.9? maybe you have pcl-1.9 but build with pcl-1.7, you can print your PCL version with adding the following command in your CMakeLists.txt. to print the version of PCL you use
``message(STATUS "***** PCL version: ${PCL_VERSION} *****")``

ziv-lin, 2019-12-10:
Besides, how large is the RAM you have? is it sufficient enough? And please kindly check Loam-Livox has the permission to write the log files and create the directories (under the dir ${home}/Loam_livox).

YangSiri, 2019-12-11:
Thanks for your suggestions, @ziv-lin 

As you can see from this, my environments
[image]
[image]

And for your second comment, i have 16GB RAM i guess. Yeah, i noticed there is a folder in _/home_ after running. Does this mean it already has the permission?
[image]

Actually, my problem comes at the _laserMapping_ part as you can see from the fig. 1,2. I also run into similar situation launching _loam\a-loam_ . I think now it is more related to my environment. But i haven't found any clue about this. Please help me with your expert experience, thank you.

And it would be great any of you can explain me about this:
[image]

Looking forward to generous help!

ziv-lin, 2019-12-12:
I am so sorry that I can not find out the problem with your environment since I have never meet the similar problems... 
Maybe you can try to find a cleaner environment to reconfigure all the settings?

whiteineyes, 2019-12-12:
@ziv-lin I reinstall Ubuntu18 system with ROS-melodic and PCL1.8, now it works fine  but crushed after loo closure just like [issue 31](https://github.com/hku-mars/loam_livox/issues/31)  ;
and I record a rosbag with my own livox mid40 in the campus of CNU which the mid40 is mounted on a experimental Electric tricycle , as you can see fig1 and fig2 , the original path is smoother than the path after loop closure , is this a potential bug ?
[image]
[image]

YangSiri, 2019-12-13:
Hi @ziv-lin @whiteineyes 

Rebuilt PCL 1.8 helps at end. Thanks

closed 2019-12-13 by YangSiri

colinlin1982, 2020-03-04:
> Hi @ziv-lin @whiteineyes
> 
> Rebuilt PCL 1.8 helps at end. Thanks

Hi @ziv-lin @whiteineyes

Would you please share your environment? I meet the same problem, my environment is:
Ubuntu 18.08 + ROS Melodic + PCL 1.9.1 + FLANN 1.9.1
First I setup the environment and run command:
roslaunch loam_livox rosbag_largescale.launch # in first terminal
rosbag play ~/catkin_ws/rosbag_for_loam_livox/CYT_02.bag # in second terminal

after a few second, the livox_lasermapping thread corrupted with:
double free or corruption(out)
BTW, I did not find any livox_lasermapping-3*.log as it mentioned.

Then I tried to re-compile everything with --std=c++17 -faligned-new, including EIGEN, PCL, loam_livox. It does not help.

Then I uninstall PCL 1.9.1, compile and install PCL-1.8.1, but when run catkin_make in ~/catkin_ws，error occurs：

-- Could NOT find Flann (missing: FLANN_INCLUDE_DIRS)
CMake Error at /usr/share/pcl-1.8/PCLConfig.cmake:45 (message):
  kdtree is required but flann was not found
Call Stack (most recent call first):
  /usr/share/pcl-1.8/PCLConfig.cmake:684 (pcl_report_not_found)
  /usr/share/pcl-1.8/PCLConfig.cmake:847 (find_external_library)
  loam_livox/CMakeLists.txt:25 (find_package)

-- Configuring incomplete, errors occurred!

I think there is something wrong with my os.

Now I'm going to reinstall my os, would you please share your environment, or further advice?