hku-mars/loam_livox #34: ScanRegistration keeps crashing.

griz1112, 2019-12-10:
While driving the robot around if I yaw too quickly ScanRegistration crashes with exit code -11. Sometimes it happens while just driving straight ahead. Is there a way to turn off loop closure ?

ziv-lin, 2019-12-12:
We will ease this case in the next update, which will be updated together with the support of "Livox mid-100".

ziv-lin, 2019-12-21:
The latest version has fix some bugs in feature extraction, can you update your code and have a try?

griz1112, 2019-12-23:
I will test in the next couple of days. One thing I found that caused the crashes was objects being too close to the laser.  Say you get too close to a wall and do a turn of 90 deg or so. I think its because it has nothing to work with for a long enough time period that it crashes.

closed 2019-12-29 by ziv-lin