hku-mars/loam_livox #35: cell's center, covariance and feature_direction

narutojxl, 2019-12-12:
Hi @ziv-lin , thanks for sharing your work to community. when calculating cell's center cell_map_keyframe.hpp func  find_cell_center()  and covariance func get_covmat() in code is as follows, different from the paper equ1. and  equ3. repectively, does this matter?
>  cell_center(0) = (std::round((pt(0) - half_of_box_size) / box_size)) * box_size + half_of_box_size;
    cell_center(1) = (std::round((pt(1) - half_of_box_size) / box_size)) * box_size + half_of_box_size;
    cell_center(2) = (std::round((pt(2) - half_of_box_size) / box_size)) * box_size + half_of_box_size;

> for (size_t i = 0; i < pt_size; i++) {
      m_cov_mat = m_cov_mat + (m_points_vec[i] * m_points_vec[i].transpose()).template cast<COMP_TYPE>();
   }
                m_cov_mat -= pt_size * (m_mean * m_mean.transpose());
                m_cov_mat /= (pt_size - 1);
 
When calculating feature direction in func feature_direction(), in paper V-C(2D histogram of keyframe),  " we choose the direction with positive X components", in code we transform points whose x part is negative into positive, could you please explain it? Looking forward to your reply, thanks for your help!
>  if (vec_3d[0] < 0) { 
           vec_3d *= (-1.0); 
        }

ziv-lin, 2019-12-12:
Since the direction in 3D space have two directions, for convenience, we choose the direction with its x value alway positive. In addition, we have mentioned this point in our preprint paper:)
[image]

"we choose the direction with positive X components, i.e., C_d = sign(C_{d_x}) · C_d."

ziv-lin, 2019-12-12:
The compute of covariance is correct, which is more effective that can save more time in computation.

narutojxl, 2019-12-13:
Hi @ziv-lin , thanks for your reply! 
For covariance calculate, i don't understand why paper equ3. equal to the format in code, could you please give some tips?
[image]
For calculate histogram of keyframe, you say "we choose the direction with positive X components, i.e., C_d = sign(C_{d_x}) · C_d.". I understand what you mean is  for negative x component, we should ignore these cells, we only use positive x of cells to calculate histogram. Did I get it wrong?
For center i see the code also slightly different from the paper equ1.

narutojxl, 2019-12-16:
There is a point p(black) in the bound of the following picture,  which constructed with the center point(red) and block size. I found that some black points doesn't belong to the cell, because they created cell's center does not equal to the red one.  Is that right?  The correspongding code is [find_cell().](https://github.com/hku-mars/loam_livox/blob/master/source/cell_map_keyframe.hpp#L716)
[image]