hku-mars/loam_livox #36: Add curr frame's points into m_keyframe_of_updating_list

narutojxl, 2019-12-12:
Hi @ziv-lin, after transformed curr's points into map, when insert these celles into `m_keyframe_of_updating_list`  called in [add_cells into each keyframe of updating_list](https://github.com/hku-mars/loam_livox/blob/master/source/laser_mapping.hpp#L1531), we only insert the first cell into each keyframe, right?  why we don't  insert all of cells into each of them?
Thanks for your help!
>
   	    FUNC_T void add_cells( const std::set< Mapping_cell_ptr > &cells_vec )
	    {
		unsigned int last_cell_numbers = 0;
		for ( typename std::set< Mapping_cell_ptr >::iterator it = cells_vec.begin(); it != cells_vec.end(); it++ )
		{
		    m_set_cell.insert( *( it ) );
		    m_map_pt_cell.insert( std::make_pair((*(it))->get_center(), *it) );
		    if(last_cell_numbers!= m_set_cell.size()) // New different cell
		    {
		        if(last_cell_numbers == 0)
		        {
		            // Init octree
		            m_pcl_cells_center->push_back( pcl::PointXYZ( (*it)->m_center( 0 ), (*it)->m_center( 1 ), (*it)->m_center( 2 ) ) );
		        }
		        last_cell_numbers = m_set_cell.size();
		    }
		};
		m_accumulate_frames++;
	    }

ziv-lin, 2019-12-12:
Not only just the first cell, we insert all of the cells which have been visited in this scan.

narutojxl, 2019-12-13:
Hi @ziv-lin,  why when we insert  cell's center into `m_pcl_cells_center` only when `last_cell_numbers == 0`, don't figure out this.

ziv-lin, 2019-12-14:
This is some of the codes of legacy, in the older version, I maintain a octree in Maps, which have beed deleted in one of the version, but I forget to delete this code :)

closed 2019-12-16 by narutojxl