hku-mars/loam_livox #37: initial trans value between curr anchor keyframe and history keyframe, before loop calculate

narutojxl, 2019-12-12:
Hi @ziv-lin, when we detect there is a potential loop, before we use different voxel resolution to filter the curr anchor keyframe and history anchor history keyframe, we set the initial trans  use keyframe's center. I guess we calculate transform is: the TF from curr to history in world frame. Here the transform direction is reffering to frame, not vectors in frame. Why we don't set the trans initial is history minus curr,

>  Eigen::Matrix< double, 3, 1 > transform_T = 
 ( keyframe_b->get_center() - keyframe_a->get_center() ).template cast< double >();

It is the vector from curr to history.  Thanks for your help very much!

ziv-lin, 2019-12-12:
This is for aligning the center of two **keyframes**, so we subtract the bias in scene alignment.

narutojxl, 2019-12-13:
@ziv-lin Thanks a lot!

closed 2019-12-13 by narutojxl