hku-mars/loam_livox #38: sometimes not trigger loop under the same params and dataset

narutojxl, 2019-12-13:
Hi @ziv-lin , i use  rosbag_loop.launch and don't modify any param in performance_precision.yaml , play with loop_hku_zym.bag.  I run the demo 5 times,  and i found sometimes  it doesn't trigger loop, but sometimes it can. Did you meet this ? Here is This is the result of a experiment  not  loop.
[image]

ziv-lin, 2019-12-13:
This is because I add some **randomness** during the alignment of the map. **For realtime consideration, we randomly** drop some of the matchings during ICP of map alignment, this might cause the performance not the alway the same of each launching. 
Once the maps can not alignment well, we treat the loop detect by 2D-histogram as false positive.

ziv-lin, 2019-12-13:
For disable such  randomness, you can modify the value in performance_precision.yaml
``loop_closure/scene_alignment_maximum_residual_block: 3000``
from 3000 to an extrmely high value (like 999999).

In addition, you can also fine-tune the value
``loop_closure/map_alignment_inlier_threshold: 0.2``
to a larger value to accept the loop.

narutojxl, 2019-12-16:
@ziv-lin, thanks a lot :)

closed 2019-12-16 by narutojxl