hku-mars/loam_livox #40: How to export/save the pointcloud from the rivz?

Huang88888888Fei, 2019-12-17:
Hi, I wonder how can I export the result to *.pcd files. Now I have recorded a rosbag file, and it looks good in rviz, but I can't get the coordinates of the points. How can I export them to *.pcd files? Then I can process them in pcl. Thank you!

dzxbiubiubiu, 2020-02-28:
Hi，I have the same question with you recently. Have you solve the question?Could you tell me how to save the pointcloud?Thank you!

Huang88888888Fei, 2020-02-28:
Please refer: https://www.guyuehome.com/860. But I did not actually test it.

Huang88888888Fei, 2020-02-28:
> Hi，I have the same question with you recently. Have you solve the question?Could you tell me how to save the pointcloud?Thank you!

Please refer: https://www.guyuehome.com/860. But I did not actually test it.

pierreramiro, 2022-12-27:
check #109