hku-mars/loam_livox #42: Why there are some points on the trajectory？

Leon1128, 2020-01-04:
Hi！I use rosbag_mid100.launch to processing data from livox-mid100. The result is better than livox-mid40. But, why there are some points on the trajectory in /velodyne_cloud_registered and /laser_cloud_surround? I checked the raw data and there are no these points. How can I delete these points?
[image]
[image]
[image]

ziv-lin, 2020-01-11:
It seems that there have some noise data sampled near the lidar? 
Are your lidar blocked by the small nearby objects?

Leon1128, 2020-01-12:
I think I found the reason. There are three points at the center of three-unit of lidar of MID-100.
[image]

closed 2020-01-13 by Leon1128