hku-mars/loam_livox #43: [Question] Livox Horizon compatibility

kirkone, 2020-01-07:
Hi,

I saw the Livox Horizon device and thought this would be a good match for this kind of software.
Before I buy such a device I want to make sure it will be compatible with this Software or at least I want to help making it compatible.

Thanks for this awesome project!

CU
KirK

ziv-lin, 2020-01-11:
I got a Livox-Horizon (evaluation version) recently, but I have some troubles in querying the point cloud from livox-ros-driver.
[image]

The point cloud form Livox-horizon seem denser than Livox-mid100, the comparison  of the point cloud are shown as follow (the sampling time is all 100ms):
Livox-mid100:
[image]

Livox-horizon:
[image]

ziv-lin, 2020-01-11:
By the way, I will try to make my package work campable with Livox-horizon in recently. Please wait with patience~

byte-deve, 2020-01-28:
Really fantastic work! How long will it take to have a compatible version? I also got one horizon and played with your method. Could you share some idea for how horizon version shall be fit?

BR

ziv-lin, 2020-01-29:
@byte-deve can you try if our package works well if your launch the mid-40 configuration directly?
```
    roslaunch loam_livox livox.launch
    roslaunch livox_ros_driver livox_lidar.launch
```
or:
```
roslaunch loam_livox rosbag.launch
rosbag play YOUR_RECORDED.bag
```
I currently stay at home  (mainland China) now.  Due to the outbreak of the **Novel Coronavirus (2019-nCoV)**, **I decided not to go back to school recently** to avoid being infected by the virus.
In addition, If you got a Livox-Horizon now, it will be very appreciating that if you can record a rosbag file for me (my email: ziv.lin.ljr@gmail.com or 730470317@qq.com)... I believe the problems of livox-ros-driver have been fixed by Livox-SDK-team.

mhc994, 2020-01-30:
@ziv-lin Thanks for openning this great work which makes me not so boring at home.
Horizon gives more points than Mid-40. The program runs slowly on my laptop and uses out my 16G memory after several minutes. But fortunately it works in general.
[image]
Here are two rosbags for you.
The path in each bag is a retangle surrounding a buiding. In the second bag there is more rotating of the lidar. 
[bag1](https://cloud.tsinghua.edu.cn/f/58ab563362bb400d91bb/?dl=1) [bag2](https://cloud.tsinghua.edu.cn/f/1a2e2d9f42ce435fac58/?dl=1)

Fk the virus.

byte-deve, 2020-01-30:
@ziv-lin I‘ve got similar dense point cloud as you uploaded above, couldn't get good result for in-house situation.
@mhc994 Thanks for your bag file! Did you try it in-house?

BR

ziv-lin, 2020-01-31:
@mhc994 Thanks for your sharing~, it seems that our package can run well with Livox-horizon (with mid40 configurations). 
Besides,  since the scanning pattern of Livox-Horizon is totally different with Livox-Mid40, I believe some of the adaptations can be made to improving the overall performance.

ziv-lin, 2020-01-31:
@byte-deve Not only Livox-horizon, both the Livox-mid40 and Livox-mid100 run comparetive worse in indoor scenarios. The reasons are list as follows:
1. In indoor-scenarios, you are more likely to meet with the case of degeneration since features like plane or edge are not as soo plenty as outdoor environment, especially in such limited FoV.
2. In indoor-scenarios, there are full of high reflective objects like the mirror, glass, floor and ect,  which cause the returning measurement of distanct not so reliable.

mhc994, 2020-01-31:
Not only the indoor scenarios, the robustness is relatively weak at the crossroads (for example the top-right corner of my path above). Looking forward to further improvements.
BTW, I noticed that the livox-horizon comes with a well calibrated BMI088 IMU. Maybe adding an IMU optimization factor will be helpful.

ziv-lin, 2020-01-31:
@mhc994 Yes, fusing data from different sensors are of great help, and this research topic (sensor-fusion) is what I am now working on.

YangSiri, 2020-10-26:
Hi, @mhc994 

May I ask how you specificaly run this rode with Horizon? I got some errors while playing the horizon bag as follows?

> [ERROR] [1603703174.273220738]: Client [/livox_scanRegistration] wants topic /livox/lidar to have datatype/md5sum [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181], but our version has [livox_ros_driver/CustomMsg/e4d6829bdfe657cb6c21a746c86b21a6]. Dropping connection.

Looking forward to your replay;>

radeonwu, 2020-10-28:
> Hi, @mhc994
> 
> May I ask how you specificaly run this rode with Horizon? I got some errors while playing the horizon bag as follows?
> 
> > roslaunch loam_livox livox.launch
> > rosbag play --clock -k ./src/livoxloam2/2020_parking_lot.bag
> 
> > [ERROR] [1603703174.273220738]: Client [/livox_scanRegistration] wants topic /livox/lidar to have datatype/md5sum [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181], but our version has [livox_ros_driver/CustomMsg/e4d6829bdfe657cb6c21a746c86b21a6]. Dropping connection.
> 
> Looking forward to your replay;>

The error message already states the issue clearly. You need convert the pointcloud message type from /livox_ros_driver/CustomMsg to /sensor_msgs/PointCloud2.
A more straightforward approach is configure the horizon ros driver parameter to directly output /sensor_msgs/PointCloud2 from the sensor.