hku-mars/loam_livox #45: livox_laserMapping crashes

finostro-amtc, 2020-01-24:
Hello, 
The livox_laserMapping node is crashing with some sort of memory issue:
```
 Error in `/home/finostro/Code/ros_ws/devel/lib/loam_livox/livox_laserMapping': double free or corruption (out): 0x00007fdcaddc1bc0 
```

Maybe you could run your code with [address sanitizer](https://github.com/google/sanitizers/wiki/AddressSanitizer) to check for memory problems?

Cheers

finostro, 2020-01-27:
Found out it was a problem with eigen alignment. Fixed it by compiling with c++17

ziv-lin, 2020-01-29:
I often meet with the problems of Eigen alignment  too (not in this version, this released version are comparative stable), the problems and its correspondence solution are not always the same...
It cause me crazy many times...

finostro, 2020-01-29:
apparently with c++17  eigen handles the alignment internally so we don't have to do anything anymore :)

musicshmily, 2020-06-22:
hello,i come cross this problem also, and i make my gcc update to 9.1 so can use c++17,but i don't know how to use it to compile the eigen. my eigen is installed by :
 sudo apt-get install libeigen3-dev
so what should i do to slove this problem?
thanks.

yiakwy-mapping-team, 2021-05-07:
@finostro @ziv-lin I checked gdb (my gcc 7.5, Eigen 3.3.7, PCL 1.8.1), the error relates to Eigen memory alignment:

```
double free or corruption (out)

Thread 1 "livox_scanRegis" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff44eb921 in __GI_abort () at abort.c:79
#2  0x00007ffff4534967 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff4661b0d "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff453b9da in malloc_printerr (str=str@entry=0x7ffff4663818 "double free or corruption (out)") at malloc.c:5342
#4  0x00007ffff4542f6a in _int_free (have_lock=0, p=0x555555989530, av=0x7ffff4896c40 <main_arena>) at malloc.c:4308
#5  __GI___libc_free (mem=0x555555989540) at malloc.c:3134
#6  0x000055555556c7c1 in Eigen::internal::aligned_free (ptr=<optimized out>) at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:177
#7  Eigen::aligned_allocator<pcl::PointXYZI>::deallocate (this=0x7fffff9decb8, p=<optimized out>) at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:763
#8  std::allocator_traits<Eigen::aligned_allocator<pcl::PointXYZI> >::deallocate (__a=..., __n=<optimized out>, __p=<optimized out>) at /usr/include/c++/7/bits/alloc
```

yiakwy-mapping-team, 2021-05-08:
changing PCL to 1.9 has solved the problem for ceres.