hku-mars/loam_livox #49: How to save the pointcloud as pcd

dzxbiubiubiu, 2020-02-28:
First, thanks for your code and bag.But when I try to transform the pointcloud to the pcd.Here is a question:
[image]
/opt/ros/kinetic/lib/pcl_ros/pointcloud_to_pcd: error while loading shared libraries: libpcl_io.so.1.7: cannot open shared object file: No such file or directory

But how could I find the libpcl_io.so.1.7 when I use the pcl1.9?

I would be very happy for yous any reply!

Egor-Morozov, 2020-04-16:
Hello !
i successfully tryed to use https://github.com/ANYbotics/point_cloud_io to save as ply 
i wrote about here https://github.com/hku-mars/loam_livox/issues/8 
unfortunatelly this node can write only ply ((

wbr, Egor

dzxbiubiubiu, 2020-04-17:
Oh! 
Thanks for your reply. I will try it later. Thank you very much!