hku-mars/loam_livox #51: Unable stitch scanned map and image not found

TezRyuga, 2020-03-06:
I used a usb camera as my web cam with usb_camera package installed. I executed the same process as given in the readme.md file on the repo. But I am unable to get a stitched map when I try to move the lidar( livox mid-40) or rotate it through an angle. Can you help? I am attaching the screenshots while running the rosbag file of the scan.
[image]

TezRyuga, 2020-03-06:
btw, I am using pcl 1.9.1

ziv-lin, 2020-04-27:
This panel is used for displaying the RGB image in some of my example bags files.
In addition, the attached txt file is the code I read the RGB image from USB camera can publish it via rostopic "/image/image".
[read_camera.cpp.txt](https://github.com/hku-mars/loam_livox/files/4538405/read_camera.cpp.txt)
 You can try to add this source file to your node so that you ren preview the image when you are sampling the LiDAR data.