hku-mars/loam_livox #53: play the given rosbag and scanRegistration crashed

GUIGUI93, 2020-03-29:
Hello, I am trying to use the given rosbag to recreate the result, I followed the readme, used the command: roslaunch loam_livox rosbag.launch  + rosbag play HKU_ZYM.bag, but the process scanRegistration was crashed right after I played the rosbag, and I got the message as followed:

[image]

I am not sure if it is the same issue as the issue #34, but I have updated to the latest version but still have this problem, please take a look, thanks a lot!

positron96, 2020-05-28:
Hey, I have the same problem, the program crashes with exit code -11. Have you found a solution?

SherlockHermes, 2020-07-17:
I have the same problem. The wrong message: [livox_scanRegistration-2] process has died [pid 14776, exit code -11

Auturle, 2020-08-07:
> I have the same problem. The wrong message: [livox_scanRegistration-2] process has died [pid 14776, exit code -11
> Does anyone know how to fix it?

I have the same problem. The wrong message also the same as yours. At last, I solute this proble through install the Livox-ros-driver. I guess may be your ros-driver install failed,. Hope can help you.

Pepsipep18, 2024-04-18:
> Hello, I am trying to use the given rosbag to recreate the result, I followed the readme, used the command: roslaunch loam_livox rosbag.launch + rosbag play HKU_ZYM.bag, but the process [...]

Hey, did you solve the problem? I have the same problem.