hku-mars/loam_livox #55: Residual and Jacobian evaluation failure

yuqingyang888, 2020-04-07:
When I was playing the loop_simple.bag with rosbag.launch, an error occurs. As well, it appears if I use other launch files. However, the rviz also shows the result.
I'd like to know whether it is a common case and whether it could affect the accuracy. And how to solve it?
Thx~
[image]

yuqingyang888, 2020-04-07:
I solved this problem by replacing pcl 1.7 with pcl 1.9.

closed 2020-04-07 by yuqingyang888

YangSiri, 2020-10-14:
Hi, @yuqingyang888 

I encountered the same problem as this. I want to know that you rebuild PCL 1.9 on top of 1.7(which is compiled with ROS) and the issue is gone?