hku-mars/loam_livox #56: How to save the result pointcloud of rosbag?

nonlinear1, 2020-04-10:


Egor-Morozov, 2020-04-16:
Hello!
You can install https://github.com/ANYbotics/point_cloud_io
This node can write every message of topic "/velodyne_cloud_registered" as ply cloud.
You have to edit rosbag.launch file (see example by link below)
After finish of:
roslaunch loam_livox rosbag.launch
rosbag play CYT_02.bag

you can open all ply (about 12500 for CYT_02) by cloudcompare and merge all ply parts.
And save result as las.
The link for example of rosbag.launch and the result - CYT_02.las = https://drive.google.com/open?id=1RhBJK3zEQTQWF9ixenswgs5YlTzAn_6p

wbr, Egor.

nonlinear1, 2020-04-16:
Thank you very much!

ns-1m, 2020-05-28:
Hi Egor,

Thanks for showing us how to get the 3D pointcloud (i.e. ply) and how to get a las file in Cloudcompare.

The las file is good. I open it using R package (lidR).  See logs below.

Noli 
 
~~~~~~~
class        : LAS (LASF v1.2)
point format : 2
memory       : 1.7 Gb 
extent       :-43.0077, 205.369, -120.722, 193.37 (xmin, xmax, ymin, ymax)
coord. ref.  : NA 
area         : 56033.64 units²
points       : 20.21 million points
density      : 360.69 points/units²
names        : X Y Z Intensity ReturnNumber NumberOfReturns ScanDirectionFlag EdgeOfFlightline Classification Synthetic_flag Keypoint_flag Withheld_flag ScanAngleRank UserData PointSourceID R G B Original cloud index 
File signature:           LASF 
File source ID:           0 
Global encoding:
 - GPS Time Type: GPS Week Time 
 - Synthetic Return Numbers: no 
 - Well Know Text: CRS is GeoTIFF 
 - Aggregate Model: false 
Project ID - GUID:        00000000-0000-0000-0000-000000000000 
Version:                  1.2
System identifier:        PDAL 
Generating software:      PDAL 2.1.0 (b47a76) 
File creation d/y:        106/2020
header size:              227 
Offset to point data:     473 
Num. var. length record:  1 
Point data format:        2 
Point data record length: 34 
Num. of point records:    20210994 
Num. of points by return: 20210994 0 0 0 0 
Scale factor X Y Z:       1e-06 1e-06 1e-06 
Offset X Y Z:             0 0 0 
min X Y Z:                -43.0077 -120.722 -8.16431 
max X Y Z:                205.369 193.37 50.0814 
Variable length records: 
   Variable length record 1 of 1 
       Description: Extra Bytes Record 
       Extra Bytes Description:
          Original cloud index:

YushengWHU, 2020-07-25:
Hello guys, I tried compiling this together with livox loam,  and it gave me an error that package pcl_ros and pcl_conversions not found.
when I put these two packages into catkin workspace, it gave me :
pcl_conversions/pcl_conversions.h: No such file or directory
When I tried install these two packages, also won't work.

So I am wondering which version of PCL are you using and have you produced a same problem?

Fernandoxjy14, 2021-09-13:
> Hello!
> You can install https://github.com/ANYbotics/point_cloud_io
> This node can write every message of topic "/velodyne_cloud_registered" as ply cloud.
> You have to edit rosbag.launch file (see [...]

Hi Egor,
Thanks for your information. Could you please show a little more details on how to merge all ply parts in Cloudcompare?

Thanks!