hku-mars/loam_livox #57: malloc(): memory corruption

kintzhao, 2020-04-14:
yhzhao@yhzhao:~/dataSets/3D$ rosbag play  mid100_example.bag  --clock

yhzhao@yhzhao:~/3d_lidar_ws$ roslaunch loam_livox rosbag_mid100.launch 
... logging to /home/yhzhao/.ros/log/63199d08-7df1-11ea-9e2a-98eecb94140d/roslaunch-yhzhao-29424.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/yhzhao/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://yhzhao:35821/

SUMMARY
========

PARAMETERS
 * /common/if_motion_deblur: 0
 * /common/if_save_to_pcd_files: 0
 * /common/if_update_mean_and_cov_incrementally: 1
 * /common/if_verbose_screen_printf: 1
 * /common/lidar_type: livox
 * /common/log_save_dir: /home/yhzhao/Loam...
 * /common/loop_save_dir: /home/yhzhao/Loam...
 * /common/maximum_parallel_thread: 1
 * /common/odom_mode: 1
 * /common/pcd_save_dir: /home/yhzhao/Loam...
 * /common/piecewise_number: 2
 * /common/threshold_cell_revisit: 2000
 * /feature_extraction/corner_curvature: 0.1
 * /feature_extraction/livox_min_dis: 0.1
 * /feature_extraction/livox_min_sigma: 7e-4
 * /feature_extraction/mapping_line_resolution: 0.1
 * /feature_extraction/mapping_plane_resolution: 0.4
 * /feature_extraction/minimum_view_angle: 5
 * /feature_extraction/scan_line: 64
 * /feature_extraction/surface_curvature: 0.005
 * /loop_closure/if_dump_keyframe_data: 0
 * /loop_closure/if_enable_loop_closure: 0
 * /loop_closure/map_alignment_if_dump_matching_result: 0
 * /loop_closure/map_alignment_inlier_threshold: 0.2
 * /loop_closure/map_alignment_maximum_icp_iteration: 5
 * /loop_closure/map_alignment_resolution: 0.1
 * /loop_closure/maximum_keyframe_in_waiting_list: 10
 * /loop_closure/minimum_keyframe_differen: 200
 * /loop_closure/minimum_similarity_linear: 0.65
 * /loop_closure/minimum_similarity_planar: 0.94
 * /loop_closure/scans_between_two_keyframe: 100
 * /loop_closure/scans_of_each_keyframe: 300
 * /loop_closure/scene_alignment_maximum_residual_block: 3000
 * /mapping/init_accumulate_frames: 50
 * /mapping/input_downsample_mode: 1
 * /mapping/matching_mode: 0
 * /mapping/max_allow_final_cost: 2.0
 * /mapping/max_allow_incre_R: 20
 * /mapping/max_allow_incre_T: 0.3
 * /mapping/maximum_histroy_buffer: 400
 * /mapping/maximum_in_fov_angle: 45
 * /mapping/maximum_mapping_buffer: 20000000
 * /mapping/maximum_pointcloud_delay_time: 1.0
 * /mapping/maximum_search_range_corner: 100
 * /mapping/maximum_search_range_surface: 100
 * /mapping/surround_pointcloud_resolution: 0.3
 * /optimization/ceres_maximum_iteration: 50
 * /optimization/icp_maximum_iteration: 15
 * /optimization/max_allow_final_cost: 2.0
 * /optimization/maximum_residual_blocks: 200
 * /optimization/minimum_icp_R_diff: 0.01
 * /optimization/minimum_icp_T_diff: 0.01
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    livox_laserMapping (loam_livox/livox_laserMapping)
    livox_scanRegistration (loam_livox/livox_scanRegistration)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[livox_scanRegistration-1]: started with pid [29441]
process[livox_laserMapping-2]: started with pid [29442]
process[rviz-3]: started with pid [29443]
========= Hello, this is livox laser ========
Compile time:  18:45:36
Softward version: V_0.1_beta
========= End ========
~~~~~ Init livox lidar parameters ~~~~~
[Features_extraction_ros_param]: common/lidar_type ==> livox
***** I get lidar_type declaration, lidar_type_name = livox ***** 
Set lidar type = livox
[Laser_mapping_ros_param]: feature_extraction/mapping_line_resolution ==> 0.1
[Features_extraction_ros_param]: feature_extraction/livox_min_dis ==> 0.1
Set livox lidar minimum distance= 0.1
[Laser_mapping_ros_param]: feature_extraction/mapping_plane_resolution ==> 0.4
[Features_extraction_ros_param]: feature_extraction/livox_min_sigma ==> 0.0007
Set livox lidar minimum sigama =  0.0007
~~~~~ End ~~~~~
[Laser_mapping_ros_param]: common/if_verbose_screen_printf ==> 0
[Features_extraction_ros_param]: feature_extraction/scan_line ==> 64
[Laser_mapping_ros_param]: common/odom_mode ==> 1
[Features_extraction_ros_param]: feature_extraction/mapping_plane_resolution ==> 0.4
[Laser_mapping_ros_param]: common/maximum_parallel_thread ==> 1
[Features_extraction_ros_param]: feature_extraction/mapping_line_resolution ==> 0.1
[Features_extraction_ros_param]: feature_extraction/minimum_range ==> 0.1
[Laser_mapping_ros_param]: common/if_motion_deblur ==> 0
[Features_extraction_ros_param]: common/if_motion_deblur ==> 0
[Laser_mapping_ros_param]: common/if_save_to_pcd_files ==> 0
[Features_extraction_ros_param]: common/piecewise_number ==> 2
[Laser_mapping_ros_param]: common/if_update_mean_and_cov_incrementally ==> 1
[Features_extraction_ros_param]: common/odom_mode ==> 1
[Laser_mapping_ros_param]: common/threshold_cell_revisit ==> 2000
[Features_extraction_ros_param]: feature_extraction/corner_curvature ==> 0.1
[Laser_mapping_ros_param]: optimization/minimum_icp_R_diff ==> 0.01
[Features_extraction_ros_param]: feature_extraction/surface_curvature ==> 0.005
[Laser_mapping_ros_param]: optimization/minimum_icp_T_diff ==> 0.01
[Features_extraction_ros_param]: feature_extraction/minimum_view_angle ==> 5
[Laser_mapping_ros_param]: optimization/icp_maximum_iteration ==> 15
[Features_extraction_ros_param]: log_save_dir ==> ../
Open scanRegistration.log successful.
[Laser_mapping_ros_param]: optimization/ceres_maximum_iteration ==> 50
[Laser_mapping_ros_param]: optimization/maximum_residual_blocks ==> 200
[Laser_mapping_ros_param]: optimization/max_allow_incre_R ==> 4
[Laser_mapping_ros_param]: optimization/max_allow_incre_T ==> 2
[Laser_mapping_ros_param]: optimization/max_allow_final_cost ==> 2
[Laser_mapping_ros_param]: mapping/init_accumulate_frames ==> 50
[Laser_mapping_ros_param]: mapping/maximum_histroy_buffer ==> 400
[Laser_mapping_ros_param]: mapping/maximum_mapping_buffer ==> 20000000
[Laser_mapping_ros_param]: mapping/matching_mode ==> 0
[Laser_mapping_ros_param]: mapping/input_downsample_mode ==> 1
[Laser_mapping_ros_param]: mapping/maximum_in_fov_angle ==> 45
[Laser_mapping_ros_param]: mapping/maximum_pointcloud_delay_time ==> 1
[Laser_mapping_ros_param]: mapping/maximum_in_fov_angle ==> 45
[Laser_mapping_ros_param]: mapping/maximum_search_range_corner ==> 100
[Laser_mapping_ros_param]: mapping/maximum_search_range_surface ==> 100
[Laser_mapping_ros_param]: mapping/surround_pointcloud_resolution ==> 0.3
[Laser_mapping_ros_param]: loop_closure/if_enable_loop_closure ==> 0
[Laser_mapping_ros_param]: loop_closure/minimum_keyframe_differen ==> 200
[Laser_mapping_ros_param]: loop_closure/minimum_similarity_linear ==> 0.65
[Laser_mapping_ros_param]: loop_closure/minimum_similarity_planar ==> 0.94
[Laser_mapping_ros_param]: loop_closure/map_alignment_resolution ==> 0.1
[Laser_mapping_ros_param]: loop_closure/map_alignment_inlier_threshold ==> 0.2
[Laser_mapping_ros_param]: loop_closure/map_alignment_maximum_icp_iteration ==> 5
[Laser_mapping_ros_param]: loop_closure/maximum_keyframe_in_waiting_list ==> 10
[Laser_mapping_ros_param]: loop_closure/scans_of_each_keyframe ==> 300
[Laser_mapping_ros_param]: loop_closure/scans_between_two_keyframe ==> 100
[Laser_mapping_ros_param]: loop_closure/scene_alignment_maximum_residual_block ==> 3000
[Laser_mapping_ros_param]: loop_closure/if_dump_keyframe_data ==> 0
[Laser_mapping_ros_param]: loop_closure/map_alignment_if_dump_matching_result ==> 0
[Laser_mapping_ros_param]: common/log_save_dir ==> /home/yhzhao/Loam_livox
Open mapping.log successful.
Open timer.log successful.
Open match_buff.log successful.
[Laser_mapping_ros_param]: common/pcd_save_dir ==> /home/yhzhao/Loam_livox
[Laser_mapping_ros_param]: common/loop_save_dir ==> /home/yhzhao/Loam_livox
Open common.log successful.
Open timer.log successful.
Open poses.log successful.
*** Error in `/home/yhzhao/3d_lidar_ws/devel/lib/loam_livox/livox_laserMapping': malloc(): memory corruption: 0x00000000024ad07f ***
livox_laserMapping: /usr/include/boost/smart_ptr/shared_ptr.hpp:648: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = const sensor_msgs::PointCloud2_<std::allocator<void> >; typename boost::detail::sp_member_access<T>::type = const sensor_msgs::PointCloud2_<std::allocator<void> >*]: Assertion `px != 0' failed.
[livox_laserMapping-2] process has died [pid 29442, exit code -6, cmd /home/yhzhao/3d_lidar_ws/devel/lib/loam_livox/livox_laserMapping __name:=livox_laserMapping __log:=/home/yhzhao/.ros/log/63199d08-7df1-11ea-9e2a-98eecb94140d/livox_laserMapping-2.log].
log file: /home/yhzhao/.ros/log/63199d08-7df1-11ea-9e2a-98eecb94140d/livox_laserMapping-2*.log
^C[rviz-3] killing on exit
[livox_scanRegistration-1] killing on exit
^Cshutting down processing monitor...
... shutting down processing monitor complete
done
yhzhao@yhzhao:~/3d_lidar_ws$ ^C

kintzhao, 2020-04-15:
I modify it by remove ptr operation

https://github.com/PointCloudCN/loam_livox

aniltotakura, 2021-07-21:
@kintzhao I tried out your fork where you modified the src code by removing pointer operation. But the usage of ram keeps on growing until there is no ram left for the system, at which point the "laser_mapping" node receives SIGKILL and gets terminated. Any idea what to do here?