hku-mars/loam_livox #58: livox_laserMapping crashed

zyjiang0571, 2020-04-24:
Hi, I am trying to use  the command: roslaunch loam_livox rosbag.launch + rosbag play HKU_ZYM.bag, but the process of livox_laserMapping  crashed right after I played the rosbag 20s or so, and I got the message as followed:
_**double free or corruption (out)
[livox_laserMapping-3] process has died [pid 9970, exit code -6, cmd /home/jiang/catkin_ws/devel/lib/loam_livox/livox_laserMapping __name:=livox_laserMapping __log:=/home/jiang/.ros/log/0ad68f4c-85ea-11ea-9c8e-001e8c74d8da/livox_laserMapping-3.log].
log file: /home/jiang/.ros/log/0ad68f4c-85ea-11ea-9c8e-001e8c74d8da/livox_laserMapping-3*.log**_

[image]

ziv-lin, 2020-04-27:
From my experience, the "double free corruption" is mainly caused by the [Eigen alignment issue](https://eigen.tuxfamily.org/dox/group__DenseMatrixManipulation__Alignement.html), and the reason is various with the different environment (e.g. the version of GCC, Eigen, PCL and etc; different compile options).
Here might be some suggestions that might help you.
1. Using C++17 compile option (refer to issue https://github.com/hku-mars/loam_livox/issues/45 ).
2. Update your Eigen, PCL.
3. Address the problem by following the solution provided by Eigen.
    a. https://eigen.tuxfamily.org/dox/group__TopicWrongStackAlignment.html
    b. https://eigen.tuxfamily.org/dox/group__TopicStructHavingEigenMembers.html

gaofei-forever, 2021-05-05:
i meet this problem too. did you have solved?  could you tell me how to solve the problem.

yiakwy-mapping-team, 2021-05-07:
I meet this problem to. in Linux docker environment + clion. gcc 7.5. Everything goes fine before start the node. My environment works very well with many driving software. We deal with eigen carefully. The Eigen version is 3.3.7, and the PCL is 1.8.1. 

Here is clion snapshot in remote docker container

```
/usr/bin/cmake -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - Unix Makefiles" /tmp/tmp.UVW2xUU27H
-- Using CATKIN_DEVEL_PREFIX: /tmp/tmp.UVW2xUU27H/cmake-build-debug/devel
-- Using CMAKE_PREFIX_PATH: /catkin_ws/devel;/WorkSpace/Github/catkin_ws/devel;/opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /tmp/tmp.UVW2xUU27H/cmake-build-debug/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- execute_process : COMMAND;/usr/bin/python2;/opt/ros/melodic/share/catkin/cmake/parse_package_xml.py;/opt/ros/melodic/share/catkin/cmake/../package.xml;/tmp/tmp.UVW2xUU27H/cmake-build-debug/catkin/catkin_generated/version/package.cmake
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- execute_process : COMMAND;/usr/bin/python2;/tmp/tmp.UVW2xUU27H/cmake-build-debug/catkin_generated/generate_cached_setup.py
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- execute_process : COMMAND;/tmp/tmp.UVW2xUU27H/cmake-build-debug/catkin_generated/env_cached.sh;/usr/bin/python2;/opt/ros/melodic/share/catkin/cmake/order_paths.py;/tmp/tmp.UVW2xUU27H/cmake-build-debug/catkin_generated/ordered_paths.cmake;--paths-to-order;/opt/ros/melodic/include;/usr/include;/opt/ros/melodic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp;/usr/include/opencv;--prefixes;/tmp/tmp.UVW2xUU27H/cmake-build-debug/devel;/opt/ros/melodic
-- execute_process : COMMAND;/tmp/tmp.UVW2xUU27H/cmake-build-debug/catkin_generated/env_cached.sh;/usr/bin/python2;/opt/ros/melodic/share/catkin/cmake/order_paths.py;/tmp/tmp.UVW2xUU27H/cmake-build-debug/catkin_generated/ordered_paths.cmake;--paths-to-order;/opt/ros/melodic/lib;--prefixes;/tmp/tmp.UVW2xUU27H/cmake-build-debug/devel;/opt/ros/melodic
-- Found required Ceres dependency: Eigen version 3.3.7 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found required Ceres dependency: gflags
-- Found Ceres version: 2.0.0 installed in: /usr/local with components: [EigenSparse, SparseLinearAlgebraLibrary, LAPACK, SuiteSparse, CXSparse, SchurSpecializations, Multithreading]
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
--   atomic
--   regex
-- Could NOT find ensenso (missing: ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** io features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing: DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** io features related to davidSDK will be disabled
-- Could NOT find DSSDK (missing: _DSSDK_LIBRARIES) 
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

** WARNING ** io features related to libusb-1.0 will be disabled
[... 19 lines cut ...]
-- looking for PCL_SEGMENTATION
-- looking for PCL_VISUALIZATION
-- looking for PCL_SURFACE
-- looking for PCL_REGISTRATION
-- looking for PCL_KEYPOINTS
-- looking for PCL_TRACKING
-- looking for PCL_RECOGNITION
-- looking for PCL_STEREO
-- looking for PCL_APPS
-- looking for PCL_IN_HAND_SCANNER
-- looking for PCL_MODELER
-- looking for PCL_POINT_CLOUD_EDITOR
-- looking for PCL_OUTOFCORE
-- looking for PCL_PEOPLE
-- ***** PCL version: 1.8.1 *****
-- execute_process : COMMAND;/usr/bin/python2;/opt/ros/melodic/share/catkin/cmake/parse_package_xml.py;/tmp/tmp.UVW2xUU27H/package.xml;/tmp/tmp.UVW2xUU27H/cmake-build-debug/catkin_generated/package.cmake
-- execute_process : COMMAND;/tmp/tmp.UVW2xUU27H/cmake-build-debug/catkin_generated/env_cached.sh;/usr/bin/python2;/usr/bin/empy;--raw-errors;-F;/tmp/tmp.UVW2xUU27H/cmake-build-debug/catkin_generated/pkg.develspace.context.pc.py;-o;/tmp/tmp.UVW2xUU27H/cmake-build-debug/devel [... cut ...] c.em
-- execute_process : COMMAND;/tmp/tmp.UVW2xUU27H/cmake-build-debug/catkin_generated/env_cached.sh;/usr/bin/python2;/usr/bin/empy;--raw-errors;-F;/tmp/tmp.UVW2xUU27H/cmake-build-debug/catkin_generated/pkg.installspace.context.pc.py;-o;/tmp/tmp.UVW2xUU27H/cmake-build-debug/catkin_generated/installspace/loam_livox.pc;/opt/ros/melodic/share/catkin/cmake/em/pkg.pc.em
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.UVW2xUU27H/cmake-build-debug

[Finished]

```

yiakwy-mapping-team, 2021-05-07:
> From my experience, the "double free corruption" is mainly caused by the [Eigen alignment issue](https://eigen.tuxfamily.org/dox/group__DenseMatrixManipulation__Alignement.html), and the reason is [...]

carsh for mapping has nothing to do with Eigen.

yiakwy-mapping-team, 2021-05-08:
updating PCL to 1.9 solved the problem:

[image]

gaofei-forever, 2021-05-09:
@yiakwy-mapping-team Did you used  pcl 1.9.0 ??

yiakwy-mapping-team, 2021-05-10:
> @yiakwy-mapping-team Did you used pcl 1.9.0 ??

Yes I have local installer to do this job. It also works for pcl-1.10.1 (with SSE patch). You can find pcl SSE patch in apollo.auto project.

Fellowflyingsky, 2022-04-07:
i have meet this for many times i dont know how to solve yet .my gcc 9.1 eigen3.37 ceres2.0 pcl 1.9
can you help me 
best thanks

Fellowflyingsky, 2022-04-07:
gcc 9.1 can allign eigen 3.37//

Fellowflyingsky, 2022-04-07:
> > @yiakwy-mapping-team Did you used pcl 1.9.0 ??
> 
> Yes I have local installer to do this job. It also works for pcl-1.10.1 (with SSE patch). You can find pcl SSE patch in apollo.auto project.

can you see my problem

Fellowflyingsky, 2022-04-07:
> From my experience, the "double free corruption" is mainly caused by the [Eigen alignment issue](https://eigen.tuxfamily.org/dox/group__DenseMatrixManipulation__Alignement.html), and the reason is various with the different environment (e.g. the version of GCC, Eigen, PCL and etc; different compile options). Here might be some suggestions that might help you.
> 
> 1. Using C++17 compile option (refer to issue [livox_laserMapping crashes #45](https://github.com/hku-mars/loam_livox/issues/45) ).
> 2. Update your Eigen, PCL.
> 3. Address the problem by following the solution provided by Eigen.
>    a. https://eigen.tuxfamily.org/dox/group__TopicWrongStackAlignment.html
>    b. https://eigen.tuxfamily.org/dox/group__TopicStructHavingEigenMembers.html

can you see my problem ，best thanks

Fellowflyingsky, 2022-04-07:
> carsh for mapping has nothing to do with Eigen.

can you see my problem ，best thanks

Fellowflyingsky, 2022-04-07:
> gcc 9.1 can allign eigen 3.37//

gcc 9.1 can allign eigen 3.37？