hku-mars/loam_livox #59: RLException: Unable to launch  livox_scanRegistration-2

kunnalparihar, 2020-04-24:
on running - "roslaunch loam_livox livox.launch"
I get following error:
RLException: Unable to launch [livox_scanRegistration-2]. 
If it is a script, you may be missing a '#!' declaration at the top.
The traceback for the exception was written to the log [file]

PARAMETERS
/rosdistro: melodic
/rosversion: 1.14.3
pcl version: 1.8

[image]

Thanks in advance

ziv-lin, 2020-04-27:
I google the problem, maybe some strangers problem cause you can not launch the package normally.
Can you:
1. Check if you compile our package correctly? can you see the binary file on folder /YOURCATKIN_WORKSPACE/devel/lib/loam_livox/
2. If you find these binary files,  if they are executable, try to "chmod +x" to them.

kunnalparihar, 2020-04-27:
Thanks a lot. Your suggestion worked.
After deleting the build and devel folder, and compiling using catkin make.
Then tried "chmod +x" to the binary file on /YOURCATKIN_WORKSPACE/devel/lib/loam_livox/ 
Now I am able to run the launch file

closed 2020-04-27 by kunnalparihar