hku-mars/loam_livox #65: catkin_make error

onionysy, 2020-06-15:
Does this algorithm relying on a particular version of Ceres?
When I make the catkin, this problem arise.
Can you give me some advice about how to handle it? THX
[image]

SherlockHermes, 2020-07-17:
I have the same problem. Maybe you should change the eigen version to 3.3.7.

guoangel, 2020-07-20:
I have another catkin_make error
[image]

YushengWHU, 2020-07-25:
Got same question here, solved by update ceres to 1.14.0(latest)

closed 2020-08-12 by onionysy

hitdezhazha, 2021-04-06:
My computer is stocked with ceres 2.0(latest),eigen 3.3.7,still not work.