hku-mars/loam_livox #67: Map Saving and Offline Localization

QingkaiLu, 2020-06-25:
We have been using loam_livox for livox mid-40 and horizon. I have two questions about loam_livox. 
1. Does loam_livox support saving the global map as a single pcd file? I tried to set [if_save_to_pcd_files](if_save_to_pcd_files) to be 1, but that doesn't generate a single pcd file of the map. 
2. Is there a way to load the global map and perform localization?

Fernandoxjy14, 2021-09-13:
> We have been using loam_livox for livox mid-40 and horizon. I have two questions about loam_livox.
>
> 1. Does loam_livox support saving the global map as a single pcd file? I tried to set [...]

Hi,
I've met the same problem. Have you figured it out? Thanks!

DormouseLi, 2022-02-15:
Same problem