hku-mars/loam_livox #69: cant reach real-time performance

YushengWHU, 2020-07-21:
Hello here, I am currently using a Huawei Matebook Xpro 2020,16GB RAM,  i7 10510u CPU and compling with pcl 1.9.1 and ceres 1.14.0.
As you've shown in your essay that the algorithm can get the real time performance with Manifold2，which is only a i7 8550u CPU, so I am wondering why I can't get the real-time processing.
By the way,  I've tested CYT and HKU_ZYM bag, with the terminal always showing the yellow label " Drop lidar frame in mapping for real time performance !!!"
XDDD, I am just a little confused, should I buy a Manifold2?

YushengWHU, 2020-07-21:
Also, I've tried a 10 times slower bag play,
rosbag play -r 0.1 xxx.bag
which works fine for my modified LEGOLOAM+GPS solutions, but for your application, it appears like this.
[image]
Do you have any suggestions?

YushengWHU, 2020-07-27:
solved everything here, that's all becaise I wanna use modified LegoLoam together with this and several other algorithm together, while LegoLoam and some others relay on PCL1.7 or lower, which has a pcl_ros and pcl_conversions conflict with PCL1.9, after I fully removed my PCL 1.7, everything's fine, and Manifold2 a really good platform, much better than my old Manifold and Pi

closed 2020-09-09 by YushengWHU

nyinyinyanlin, 2020-09-11:
@YushengWHU did you succeed in integrating Livox LiDAR with LegoLOAM+GPS?

YushengWHU, 2020-09-12:
> @YushengWHU did you succeed in integrating Livox LiDAR with LegoLOAM+GPS?

I have successfully integrated Lego with GPS using a 40 lines lidar, but I didnt change it to livox yet, cause I find loam_livox and loam_horizon usable.

nyinyinyanlin, 2020-09-17:
@YushengWHU thank you. I am much eager to know if it is possible to integrate Livox with external IMU+GPS. Please hit me when you figured out a way. 

Currently, I am having problem with after loop closure point cloud. It is so sparse, given the raw cloud is much detailed, despise of the noise points. Do you happen to know any way to increase the density or resolution of the after loop closure point cloud? Thank you very much.

mameesie, 2022-01-21:
@YushengWHU How did you remove PCL 1.7? After removing i still have performance issues. (i removed PCL by sudo apt remove libpcl-dev) Did you maybe manually link the PCL 1.9. to the livox_ros_driver? Or took some other extra steps? I would be really grateful if you could help me out

YushengWHU, 2022-01-22:
> @YushengWHU How did you remove PCL 1.7? After removing i still have performance issues and i cant figure out why. (i removed PCL by sudo apt remove libpcl-dev) Did you maybe manually link the PCL 1.9. to the livox_ros_driver or took some other extra steps? I would be really grateful if you could help me out

First you gonna need sudo apt remove libpcl-dev, then you can use locate or find to find all the pcl files, and use sudo rm to remove all of them. Use sudo updatedb after deleting all PCL 1.7 related files, and use locate libpcl to check whether you have deleted all of them.

mameesie, 2022-01-23:
Thanks for your very quick reply! this definitely improved the performance.

Zheng-jia-shuo, 2022-07-16:
@YushengWHU  I want to use Ceres to optimize loam livox, but there is an error(the same as yours,i don't think it's pcl version).I will be grateful if you can help me solve it.

YushengWHU, 2022-07-16:
> @YushengWHU I want to use Ceres to optimize loam livox, but there is an error(the same as yours,i don't think it's pcl version).I will be grateful if you can help me solve it.
可以换个包试试

Zheng-jia-shuo, 2022-07-16:
@YushengWHU  还是不可以