hku-mars/loam_livox #73: 'Catkin_make' give me 'Invoking "make -j12 -l12" failed'

sylvain0, 2020-08-17:
Hi,
"catkin_make" from [2-build](https://github.com/hku-mars/loam_livox#2-build) in the readme.md give me errors. 
Does anyone have a research lead please?

NB : The Ceres and PCL installations did not pose a problem upstream

```
sylvain@GS65:~/catkin_ws$ catkin_make
Base path: /home/sylvain/catkin_ws
Source space: /home/sylvain/catkin_ws/src
Build space: /home/sylvain/catkin_ws/build
Devel space: /home/sylvain/catkin_ws/devel
Install space: /home/sylvain/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/sylvain/catkin_ws/build"
####
####
#### Running command: "make -j12 -l12" in "/home/sylvain/catkin_ws/build"
####
[ 33%] Built target read_camera
[ 50%] Building CXX object loam_livox/CMakeFiles/livox_scanRegistration.dir/source/laser_feature_extractor.cpp.o
[ 66%] Building CXX object loam_livox/CMakeFiles/livox_laserMapping.dir/source/laser_mapping.cpp.o
In file included from /home/sylvain/catkin_ws/src/loam_livox/source/laser_feature_extractor.hpp:47:0,
                 from /home/sylvain/catkin_ws/src/loam_livox/source/laser_feature_extractor.cpp:37:
/opt/ros/melodic/include/pcl_conversions/pcl_conversions.h: In function ‘void pcl_conversions::fromPCL(const pcl::Vertices&, pcl_msgs::Vertices&)’:
/opt/ros/melodic/include/pcl_conversions/pcl_conversions.h:356:30: error: no match for ‘operator=’ (operand types are ‘pcl_msgs::Vertices_<std::allocator<void> >::_vertices_type {aka std::vector<unsigned int>}’ and ‘const Indices {aka const std::vector<int>}’)
     vert.vertices = pcl_vert.vertices;
                              ^~~~~~~~
In file included from /usr/include/c++/7/vector:69:0,
                 from /opt/ros/melodic/include/nav_msgs/Odometry.h:10,
                 from /home/sylvain/catkin_ws/src/loam_livox/source/laser_feature_extractor.hpp:41,
                 from /home/sylvain/catkin_ws/src/loam_livox/source/laser_feature_extractor.cpp:37:
/usr/include/c++/7/bits/vector.tcc:179:5: note: candidate: std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(const std::vector<_Tp, _Alloc>&) [with _Tp = unsigned int; _Alloc = std::allocator<unsigned int>]
     vector<_Tp, _Alloc>::
     ^~~~~~~~~~~~~~~~~~~
/usr/include/c++/7/bits/vector.tcc:179:5: note:   no known conversion for argument 1 from ‘const Indices {aka const std::vector<int>}’ to ‘const std::vector<unsigned int>&’
In file included from /usr/include/c++/7/vector:64:0,
                 from /opt/ros/melodic/include/nav_msgs/Odometry.h:10,
                 from /home/sylvain/catkin_ws/src/loam_livox/source/laser_feature_extractor.hpp:41,
                 from /home/sylvain/catkin_ws/src/loam_livox/source/laser_feature_extractor.cpp:37:
/usr/include/c++/7/bits/stl_vector.h:461:7: note: candidate: std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::vector<_Tp, _Alloc>&&) [with _Tp = unsigned int; _Alloc = std::allocator<unsigned int>]
       operator=(vector&& __x) noexcept(_Alloc_traits::_S_nothrow_move())
       ^~~~~~~~
/usr/include/c++/7/bits/stl_vector.h:461:7: note:   no known conversion for argument 1 from ‘const Indices {aka const std::vector<int>}’ to ‘std::vector<unsigned int>&&’
/usr/include/c++/7/bits/stl_vector.h:482:7: note: candidate: std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::initializer_list<_Tp>) [with _Tp = unsigned int; _Alloc = std::allocator<unsigned int>]
       operator=(initializer_list<value_type> __l)
       ^~~~~~~~
/usr/include/c++/7/bits/stl_vector.h:482:7: note:   no known conversion for argument 1 from ‘const Indices {aka const std::vector<int>}’ to ‘std::initializer_list<unsigned int>’
In file included from /home/sylvain/catkin_ws/src/loam_livox/source/laser_feature_extractor.hpp:47:0,
                 from /home/sylvain/catkin_ws/src/loam_livox/source/laser_feature_extractor.cpp:37:
/opt/ros/melodic/include/pcl_conversions/pcl_conversions.h: In function ‘void pcl_conversions::moveFromPCL(pcl::Vertices&, pcl_msgs::Vertices&)’:
/opt/ros/melodic/include/pcl_conversions/pcl_conversions.h:373:41: error: no matching function for call to ‘std::vector<unsigned int>::swap(pcl::Indices&)’
     vert.vertices.swap(pcl_vert.vertices);
                                         ^
In file included from /usr/include/c++/7/vector:64:0,
                 from /opt/ros/melodic/include/nav_msgs/Odometry.h:10,
                 from /home/sylvain/catkin_ws/src/loam_livox/source/laser_feature_extractor.hpp:41,
                 from /home/sylvain/catkin_ws/src/loam_livox/source/laser_feature_extractor.cpp:37:
/usr/include/c++/7/bits/stl_vector.h:1229:7: note: candidate: void std::vector<_Tp, _Alloc>::swap(std::vector<_Tp, _Alloc>&) [with _Tp = unsigned int; _Alloc = std::allocator<unsigned int>]
       swap(vector& __x) _GLIBCXX_NOEXCEPT
       ^~~~
/usr/include/c++/7/bits/stl_vector.h:1229:7: note:   no known conversion for argument 1 from ‘pcl::Indices {aka std::vector<int>}’ to ‘std::vector<unsigned int>&’
In file included from /home/sylvain/catkin_ws/src/loam_livox/source/laser_feature_extractor.hpp:47:0,
                 from /home/sylvain/catkin_ws/src/loam_livox/source/laser_feature_extractor.cpp:37:
/opt/ros/melodic/include/pcl_conversions/pcl_conversions.h: In function ‘void pcl_conversions::toPCL(const Vertices&, pcl::Vertices&)’:
/opt/ros/melodic/include/pcl_conversions/pcl_conversions.h:390:30: error: no match for ‘operator=’ (operand types are ‘pcl::Indices {aka std::vector<int>}’ and ‘const _vertices_type {aka const std::vector<unsigned int>}’)
     pcl_vert.vertices = vert.vertices;
                              ^~~~~~~~
In file included from /usr/include/c++/7/vector:69:0,
                 from /opt/ros/melodic/include/nav_msgs/Odometry.h:10,
[... 186 lines cut ...]
/home/sylvain/catkin_ws/src/loam_livox/source/ceres_icp.hpp:231:13: note: use ‘-faligned-new’ to enable C++17 over-aligned new support
/home/sylvain/catkin_ws/src/loam_livox/source/ceres_icp.hpp: In instantiation of ‘static ceres::CostFunction* ceres_icp_point2plane<_T>::Create(const Eigen::Matrix<_Scalar, 3, 1>&, const Eigen::Matrix<_Scalar, 3, 1>&, const Eigen::Matrix<_Scalar, 3, 1>&, const Eigen::Matrix<_Scalar, 3, 1>&, Eigen::Matrix<_Scalar, 4, 1>, Eigen::Matrix<_Scalar, 3, 1>) [with _T = double]’:
/home/sylvain/catkin_ws/src/loam_livox/source/point_cloud_registration.hpp:414:80:   required from here
/home/sylvain/catkin_ws/src/loam_livox/source/ceres_icp.hpp:378:13: warning: ‘new’ of type ‘ceres_icp_point2plane<double>’ with extended alignment 32 [-Waligned-new=]
             new ceres_icp_point2plane( current_pt, target_line_a, target_line_b, target_line_c,  q_last, t_last ) ) );
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/sylvain/catkin_ws/src/loam_livox/source/ceres_icp.hpp:378:13: note: uses ‘void* operator new(std::size_t)’, which does not have an alignment parameter
/home/sylvain/catkin_ws/src/loam_livox/source/ceres_icp.hpp:378:13: note: use ‘-faligned-new’ to enable C++17 over-aligned new support
In file included from /home/sylvain/catkin_ws/src/loam_livox/source/laser_mapping.hpp:65:0,
                 from /home/sylvain/catkin_ws/src/loam_livox/source/laser_mapping.cpp:37:
/home/sylvain/catkin_ws/src/loam_livox/source/cell_map_keyframe.hpp: In instantiation of ‘std::__cxx11::string Points_cloud_cell<DATA_TYPE>::to_json_string() [with DATA_TYPE = float; std::__cxx11::string = std::__cxx11::basic_string<char>]’:
/home/sylvain/catkin_ws/src/loam_livox/source/cell_map_keyframe.hpp:1511:19:   required from ‘std::__cxx11::string Maps_keyframe<DATA_TYPE>::to_json_string(int&) [with DATA_TYPE = float; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/home/sylvain/catkin_ws/src/loam_livox/source/cell_map_keyframe.hpp:1543:34:   required from ‘void Maps_keyframe<DATA_TYPE>::save_to_file(const string&, const string&) [with DATA_TYPE = float; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/home/sylvain/catkin_ws/src/loam_livox/source/laser_mapping.hpp:975:98:   required from here
/home/sylvain/catkin_ws/src/loam_livox/source/cell_map_keyframe.hpp:150:33: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         for ( unsigned i = 0; i < point_number; i++ )
                               ~~^~~~~~~~~~~~~~
loam_livox/CMakeFiles/livox_laserMapping.dir/build.make:62: recipe for target 'loam_livox/CMakeFiles/livox_laserMapping.dir/source/laser_mapping.cpp.o' failed
make[2]: *** [loam_livox/CMakeFiles/livox_laserMapping.dir/source/laser_mapping.cpp.o] Error 1
CMakeFiles/Makefile2:1523: recipe for target 'loam_livox/CMakeFiles/livox_laserMapping.dir/all' failed
make[1]: *** [loam_livox/CMakeFiles/livox_laserMapping.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
```

Doublehes, 2020-09-11:
i met the same problem as yours, have you solved it yet?

sylvain0, 2020-09-11:
No 😐 I will share here if I fix this.

sylvain0, 2020-09-26:
Is this just a path error?

sylvain0, 2020-09-28:
I have upgraded to PCL 1.9 and it works

closed 2020-09-28 by sylvain0

qianxingdu, 2021-04-09:
Hello, my pcl version is 1.9.0, but I still met this problem, what's the previous pcl version of yours?
Do you have any advice for me, thank you very much!

qianxingdu, 2021-04-13:
Finally, I found that my error is about the ceres. My ceres version is 2.0.0, this version is too high, and when I degraded it to 1.14.0, I succeed!