hku-mars/loam_livox #74: How to use loam_livox for other solid state lidar? Such as Leishen CH32

bage131, 2020-08-28:
Thank you for open source about loam_livox.I'm very happy to see the method for solid state lidar.I have used it for Leishen CH32,one kind of solid state lidar,but it failed.I recorded a rosbag around the road,then played the rosbag with rosbag_loop_simple.launch.It only shifted in  one place,it didn't go forward and the result is shown in the picture.COuld you give some suggestions?What parameters should I change?Thank you very much.
[image]

FengJianLL, 2021-07-20:
I met the same problem. My device is innovusion. If you have sovled this problem, please give some suggestion. Thank you very much!

bage131, 2021-07-31:
> I met the same problem. My device is innovusion. If you have sovled this problem, please give some suggestion. Thank you very much!

I have solved it by  hdl_graph_slam .So try to use other methods.