hku-mars/loam_livox #77: Workflow using Mid-40

nyinyinyanlin, 2020-09-11:
Thank you very much for your amazing work. I have recently bought a Mid-40 and going to use it with your software for mapping purpose.

As far as I have observed from this repository, this is my understanding with typical workflow so far.

1. Install Livox SDK
2. Run Livox-ROS-Driver which connects with the LiDAR hardware and publish PointCloud2 data to ROS nodes
3. LOAM_Livox subscribes to the data stream and does it work.
4. Data can be saved into PLY format by using pointcloud compare io.

My condition is that we have a workstation setup with i7 10th Gen and 64GB RAM, but we don't really have powerful laptop to do the real time laser scanning and processing.

- As far as I understand, Livox-ROS-Driver can convert LVX files recorded from Livox Viewer into rosbags which in turn can be played back to LOAM_Livox to create the map. Is this possible? If this is possible, we are going to use a regular laptop to record LVX files and workstation to create maps.

- I tested real time scanning indoor and in RVIZ, scanned points go away so fast. I have seen someone is suggested to change the decay time to a big number for visualization. Which particular message topics are to be changed and set, so that I can see current point clouds, loam mapped point clouds, camera movement trail and pose like in the demo videos you have provided?

- When using livox.launch, LOG/PCD/Loop save directories and printed out. But when I closed the RVIZ and LOAM_Livox, only *.LOG files are left in those directories. How do I launch the program so that my scan data are saved in those directories?

- What differences do those launch files have? For example, normal launches, large scale, loop, loop simple and realtime. Are which launch settings recommended for which scenarios? 

Thank you very much for your great work, again!

nyinyinyanlin, 2020-09-17:
I have figured out, myself. Thank you.

closed 2020-09-17 by nyinyinyanlin

AliveForU, 2020-11-23:
> Thank you very much for your amazing work. I have recently bought a Mid-40 and going to use it with your software for mapping purpose.
>
> As far as I have observed from this repository, this is my [...]

Hi,I have question about how to save pacd/plt file format after loam_livox worked,I can only see pointcloud data in rviz,but don't konw how to save them.Thanks

nyinyinyanlin, 2020-11-23:
@AliveForU Please see this #8 for your purpose. Best.