hku-mars/loam_livox #84: undefined reference to `cv::Mat::updateContinuityFlag()'

panmengjia, 2021-03-26:
系统： ubuntu 18.04 
 ros版本： Melodic

我在编译该仓库的时候，始终遇到下面问题：

`/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here
[100%] Linking CXX executable /home/pmj/catkin_ws/devel/lib/loam_livox/livox_laserMapping
CMakeFiles/livox_laserMapping.dir/source/laser_mapping.cpp.o: In function `void cv::eigen2cv<float, -1, -1, 0, -1, -1>(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, cv::Mat&)':
/usr/local/include/opencv2/core/mat.inl.hpp:521: undefined reference to `cv::Mat::updateContinuityFlag()'
CMakeFiles/livox_laserMapping.dir/source/laser_mapping.cpp.o: In function `Maps_keyframe<float>::apply_guassian_blur(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, int, float)':
/usr/local/include/opencv2/core/mat.inl.hpp:521: undefined reference to `cv::Mat::updateContinuityFlag()'
collect2: error: ld returned 1 exit status
loam_livox/CMakeFiles/livox_laserMapping.dir/build.make:439: recipe for target '/home/pmj/catkin_ws/devel/lib/loam_livox/livox_laserMapping' failed
make[2]: *** [/home/pmj/catkin_ws/devel/lib/loam_livox/livox_laserMapping] Error 1
CMakeFiles/Makefile2:1523: recipe for target 'loam_livox/CMakeFiles/livox_laserMapping.dir/all' failed
make[1]: *** [loam_livox/CMakeFiles/livox_laserMapping.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

我在网上查到如下解决方案：
修改 find_package(OpenCV 3.4.2 REQUIRED)，我已经编译过该版本的opencv，并默认安装路径在/usr/local,但依然报上面的错误。

`

CrisGao, 2021-04-21:
The livox_laserMapping miss linked to Opencv library， just appened the `${OpenCV_LIBS} `to the `target_link_libraries`,liked below:
`target_link_libraries(livox_laserMapping ${catkin_LIBRARIES} ${PCL_LIBRARIES} ${CERES_LIBRARIES} ${OpenCV_LIBS})`

panmengjia, 2021-04-21:
It is caused by the problem of the opencv version.
Opencv3.2 is pre-installed on my linux system, but other people say that this version is missing this function`cv::Mat::updateContinuityFlag()`.
Then I modified the configuration file of cv_bridge and the problem was solved，as followed:
```bash
if(NOT "include;/usr/local/include;/usr/local/include/opencv2 " STREQUAL " ")
  set(cv_bridge_INCLUDE_DIRS "")
  set(_include_dirs "include;/usr/local/include;/usr/local/include/opencv2")

set(libraries "cv_bridge;/usr/local/lib/libopencv_world.so;/usr/local/lib/libopencv_world.so.3.3;/usr/local/lib/libopencv_world.so.3.3.1")
```

closed 2021-04-21 by panmengjia