hku-mars/loam_livox #89: Requirement of PC hardware

GrisaiaTrigger, 2021-05-15:
Hi, Dr. Lin. I want to ask about the requirement of computer configuration to run loam_livox. It's less than five frames in few minutes later When I use my surface pro 5 (without discrete graphic card, in vmware workstation) connected mid-40 to map outdoor. Process becomes slow , and it crashs sometimes, when I get to the corner. It also happens when I run rosbag file. Is it because of my computer performance or other causes?
Version of libraries: PCL 1.11.1，gcc 9，Eigen 3.3.9
The error code follows:
`[livox_laserMapping-3] process has died [pid 36185, exit code -9, cmd /home/dr/catkin_ws/devel/lib/loam_livox/livox_laserMapping __name:=livox_laserMapping __log:=/root/.ros/log/47df5c2c-b314-11eb-8517-000c293de1b2/livox_laserMapping-3.log].`
`log file: /root/.ros/log/47df5c2c-b314-11eb-8517-000c293de1b2/livox_laserMapping-3*.log`
Thank you!

hansgrammer, 2021-08-30:
Hello, have you solved this problem? I often encounter this problem. When I use a package of rosbage, loam_ Livox will crash directly, while map creation will continue to crash for a while

GrisaiaTrigger, 2021-11-29:
> Hello, have you solved this problem? I often encounter this problem. When I use a package of rosbage, loam_ Livox will crash directly, while map creation will continue to crash for a while

I'm sorry to answer you so later. I think this issue maybe the deficiency of RAM memory causes. It seems to be solved when I improve my swap space into 32G. But there is a new problem that the fps of the point cloud render is to low in rviz. I'm still not sure the reason and solution. 
I hope my experience may be useful to you.