hku-mars/loam_livox #92: Undefined referrence to cv::Mat::Mat() when compiling with Opencv major version 3.4.15

aniltotakura, 2021-07-12:
I have installed Open Cv from source (version 3.4) and set the Open_CV include directory to the path containing OpenCVConfig.cmake file
Here is the ouptut of the catkin_make command:
@ubuntu:/mnt/sharedVM/LOAM_LIVOX$ catkin_make
Base path: /mnt/sharedVM/LOAM_LIVOX
Source space: /mnt/sharedVM/LOAM_LIVOX/src
Build space: /mnt/sharedVM/LOAM_LIVOX/build
Devel space: /mnt/sharedVM/LOAM_LIVOX/devel
Install space: /mnt/sharedVM/LOAM_LIVOX/install
####
#### Running command: "make cmake_check_build_system" in "/mnt/sharedVM/LOAM_LIVOX/build"
####
-- Using CATKIN_DEVEL_PREFIX: /mnt/sharedVM/LOAM_LIVOX/devel
-- Using CMAKE_PREFIX_PATH: /mnt/sharedVM/LOAM_LIVOX/devel;/opt/gssi/lgprgps;/opt/ros/melodic
-- This workspace overlays: /mnt/sharedVM/LOAM_LIVOX/devel;/opt/gssi/lgprgps;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /mnt/sharedVM/LOAM_LIVOX/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - loam_livox
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'loam_livox'
-- ==> add_subdirectory(loam_livox)
CMake Error at loam_livox/CMakeLists.txt:7 (seset):
  Unknown CMake command "seset".

-- Configuring incomplete, errors occurred!
See also "/mnt/sharedVM/LOAM_LIVOX/build/CMakeFiles/CMakeOutput.log".
See also "/mnt/sharedVM/LOAM_LIVOX/build/CMakeFiles/CMakeError.log".
Makefile:1216: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
anilkumartotakura@ubuntu:/mnt/sharedVM/LOAM_LIVOX$ catkin_make
Base path: /mnt/sharedVM/LOAM_LIVOX
Source space: /mnt/sharedVM/LOAM_LIVOX/src
Build space: /mnt/sharedVM/LOAM_LIVOX/build
Devel space: /mnt/sharedVM/LOAM_LIVOX/devel
Install space: /mnt/sharedVM/LOAM_LIVOX/install
####
#### Running command: "make cmake_check_build_system" in "/mnt/sharedVM/LOAM_LIVOX/build"
####
-- Using CATKIN_DEVEL_PREFIX: /mnt/sharedVM/LOAM_LIVOX/devel
-- Using CMAKE_PREFIX_PATH: /mnt/sharedVM/LOAM_LIVOX/devel;/opt/gssi/lgprgps;/opt/ros/melodic
-- This workspace overlays: /mnt/sharedVM/LOAM_LIVOX/devel;/opt/gssi/lgprgps;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /mnt/sharedVM/LOAM_LIVOX/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... 159 lines cut ...]
                 from /mnt/sharedVM/LOAM_LIVOX/src/loam_livox/source/laser_mapping.cpp:37:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function ‘void __static_initialization_and_destruction_0(int, int)’:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: ‘pcl::SAC_SAMPLE_SIZE’ is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
   SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
   ^~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here
[ 87%] Linking CXX executable /mnt/sharedVM/LOAM_LIVOX/devel/lib/loam_livox/livox_scanRegistration
[ 87%] Built target livox_scanRegistration
[100%] Linking CXX executable /mnt/sharedVM/LOAM_LIVOX/devel/lib/loam_livox/livox_laserMapping
CMakeFiles/livox_laserMapping.dir/source/laser_mapping.cpp.o: In function `Maps_keyframe<float>::max_similiarity_of_two_image_opencv(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, int)':
/mnt/sharedVM/LOAM_LIVOX/src/loam_livox/source/cell_map_keyframe.hpp:1214: undefined reference to `cv::Mat::Mat()'
/mnt/sharedVM/LOAM_LIVOX/src/loam_livox/source/cell_map_keyframe.hpp:1214: undefined reference to `cv::Mat::Mat()'
/mnt/sharedVM/LOAM_LIVOX/src/loam_livox/source/cell_map_keyframe.hpp:1220: undefined reference to `cv::Mat::Mat()'
/mnt/sharedVM/LOAM_LIVOX/src/loam_livox/source/cell_map_keyframe.hpp:1226: undefined reference to `cv::Mat::Mat()'
CMakeFiles/livox_laserMapping.dir/source/laser_mapping.cpp.o: In function `Maps_keyframe<float>::max_similiarity_of_two_image(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, float)':
/mnt/sharedVM/LOAM_LIVOX/src/loam_livox/source/cell_map_keyframe.hpp:1172: undefined reference to `cv::Mat::Mat()'
CMakeFiles/livox_laserMapping.dir/source/laser_mapping.cpp.o:/mnt/sharedVM/LOAM_LIVOX/src/loam_livox/source/cell_map_keyframe.hpp:1172: more undefined references to `cv::Mat::Mat()' follow
collect2: error: ld returned 1 exit status
loam_livox/CMakeFiles/livox_laserMapping.dir/build.make:421: recipe for target '/mnt/sharedVM/LOAM_LIVOX/devel/lib/loam_livox/livox_laserMapping' failed
make[2]: *** [/mnt/sharedVM/LOAM_LIVOX/devel/lib/loam_livox/livox_laserMapping] Error 1
CMakeFiles/Makefile2:1562: recipe for target 'loam_livox/CMakeFiles/livox_laserMapping.dir/all' failed
make[1]: *** [loam_livox/CMakeFiles/livox_laserMapping.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j6 -l6" failed

aniltotakura, 2021-07-14:
Solved it by specifying the directory of the OpenCVCmake.config file which is basically saying to use the opencv version used by cv_bridge

closed 2021-07-14 by aniltotakura

op114129, 2022-05-30:
> 
could u tell me detailed files ,i am in trouble ,

tungtungyan, 2022-08-04:
> > 
> 
> could u tell me detailed files ,i am in trouble ,

I have the same problem. Did you solve it?

LiuL0ngl0ng, 2023-10-11:
The livox_laserMapping miss linked to Opencv library in the CMakeLists.txt， just appened the ${OpenCV_LIBS} to the target_link_libraries,liked below:
target_link_libraries(livox_laserMapping ${catkin_LIBRARIES} ${PCL_LIBRARIES} ${CERES_LIBRARIES} ${OpenCV_LIBS})