hku-mars/loam_livox #105: 请问跑一会儿就卡死退出了，而且只对着一个方向是咋回事

DoubleCcc, 2021-11-18:
请问跑一会儿就卡死退出了，而且只对着一个方向是咋回事

lifangyan16, 2022-01-08:
同问

lifangyan16, 2022-01-10:
when i run “roslaunch loam_livox livox.launch”,the pointcloud is often broken up, and just in a centrum.
it reply issue like this:
`QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
WARNING: Logging before InitGoogleLogging() is written to STDERR
W0108 11:32:59.446646 19526 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W0108 11:32:59.446843 19526 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.

lifangyan16, 2022-01-10:
ask this question in detail

ErvinZzz, 2022-01-22:
同问

DormouseLi, 2022-02-09:
same question

lifangyan16, 2022-02-11:
Uninstall and reinstall Ubuntu 18 or 20. Don't install opencv and use opencv 3 or 4 built by system.

kanglang123, 2022-07-26:
> 

Do you solve it successfully by this method?

kanglang123, 2022-07-26:
> Uninstall and reinstall Ubuntu 18 or 20. Don't install opencv and use opencv 3 or 4 built by system.

Do you solve it successfully by this method?

lifangyan16, 2022-07-26:
> > Uninstall and reinstall Ubuntu 18 or 20. Don't install opencv and use opencv 3 or 4 built by system.
> 
> Do you solve it successfully by this method?

Yes. I used to install opencv myself, and there was a problem after that. After the installation of ubuntu18 or 20, opencv comes with it, and you don't need to install it yourself. It's easy to have problems if you install opencv yourself.

kanglang123, 2022-07-26:
> > > Uninstall and reinstall Ubuntu 18 or 20. Don't install opencv and use opencv 3 or 4 built by system.
> > 
> > 
> > Do you solve it successfully by this method?
> 
> Yes. I used to install opencv myself, and there came a problem after that. After the installation of ubuntu18 or 20, opencv comes with it, and you don't need to install it yourself. It's easy to have problems if you install opencv yourself.

I check my opencv version, it is 3.2.0. I am hard to ensure this problem that's caused by it.

lifangyan16, 2022-07-26:
> 

I don't know if you have installed opencv separately after installing Ubuntu. If it is, then it's probably the problem I said, if you haven't installed opencv yourself, it should be another problem.

LiSi0205, 2023-03-02:
I haven’t installed opencv separately, but I use ubuntu16.04. Every time I use the rviz command to run other people’s .bag files or record my own bag files, I get stuck, with a black screen. My laptop is configured with i7- 12700h 3060 graphics card. I don't know if it's a hardware problem or what? Has anyone solved it yet?
我没有单独装过opencv，用的是ubuntu16.04，每次使用rviz命令进行跑其他人的.bag文件或者录自己的bag文件，都很卡，直接黑屏那种， 自己的笔记本电脑配置i7-12700h  3060的显卡。我不知道是不是硬件的问题还是什么问题？有人解决了吗？

lifangyan16, 2023-03-02:
> I haven’t installed opencv separately, but I use ubuntu16.04. Every time I use the rviz command to run other people’s .bag files or record my own bag files, I get stuck, with a black screen. My [...]

没有别的报错的话，应该是硬件的问题，用free -m命令试试看看内存

lifangyan16, 2023-03-02:
> when i run “roslaunch loam_livox livox.launch”,the pointcloud is often broken up, and just in a centrum. it reply issue like this: `QObject::connect: Cannot queue arguments of type 'QVector' (Make sure 'QVector' is registered using qRegisterMetaType().) QObject::connect: Cannot queue arguments of type 'QVector' (Make sure 'QVector' is registered using qRegisterMetaType().) WARNING: Logging before InitGoogleLogging() is written to STDERR W0108 11:32:59.446646 19526 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization. W0108 11:32:59.446843 19526 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.

'QVector' is registered ... should be omited. The 'Warnning' just says there is a mistake. These can't be used to judge where's wrong.

LiSi0205, 2023-03-02:
> > Opencv，我我我用的的是的的不知道问题还是？？有人人了吗？？我我装装装装装装装装过装过装过装过过过过过12700H 3060的的的的的的的的的的显卡显卡我不不知道知道是是是不不硬件
> 
> 没有其他的报错的话，应该是硬件的问题，用free -m命令试看看内存

我实在虚拟机上进行操作的，给虚拟机16个处理器，运行内存也是16G，应该不是内存的问题。目前使用自己录制bag的方法（roslaunch livox_ros_driver livox_lidar_rviz.launch之后rosbag record -a）很卡。录制效果也不是很理想；使用实时建图的话基本上进入两三秒钟就会卡死，rviz视野上也只是一帧点云。不知道该怎么解决了。目前准备换一个ubuntu18.04继续试试，如果有人知道怎么解决，请告知。万分感谢。