hku-mars/loam_livox #113: livox_laserMapping crashed

Fellowflyingsky, 2022-04-08:
[image]
i have meet this for many times i dont know how to solve yet .my gcc 9.1 eigen3.37 ceres2.0 pcl 1.9
can you help me
best thanks

Fellowflyingsky, 2022-04-09:
i have solved the problem . the pcl should be 1.8 
[image]

Fellowflyingsky, 2022-04-09:
[image]

LiSi0205, 2023-03-04:
> 我已经解决了这个问题。我的系统是ubuntu18.04。环境：pcl 1.8.1，opencv 3.3，ceres 1.14，eigen 3.3.7 [image] [image] [image]

我是用您的方法进行了测试，18.04+pcl1.8+opencv3.3+eigen3.3.7+ceres1.14.0  但是在运行demo的过程中只有路径，并看不到点云，是哪里出问题了吗?
[image]