hku-mars/loam_livox #114: How to understand this part of code?

wangx1996, 2022-04-09:
in livox_feature_extractor.cpp 497 colum

`                   // TODO: handle this case.
                    screen_out << "First point should be normal!!!" << std::endl;

                    pt_info->pt_2d_img << 0.01, 0.01;
                    pt_info->polar_dis_sq2 = 0.0001;
                    add_mask_of_point( pt_info, e_pt_000 );
                    //return 0;`

and column 588

`                    if ( m_pts_info_vec[ split_idx[ val_index + 1 ] ].polar_dis_sq2 > 10000 )
                    {
                        pt_angle_index = split_idx[ val_index + 1 ] - ( int ) ( internal_size * 0.20 );
                        scan_angle = atan2( m_pts_info_vec[ pt_angle_index ].pt_2d_img( 1 ), m_pts_info_vec[ pt_angle_index ].pt_2d_img( 0 ) ) * 57.3;
                        scan_angle = scan_angle + 180.0;
                    }
                    else
                    {
                        pt_angle_index = split_idx[ val_index + 1 ] - ( int ) ( internal_size * 0.80 );
                        scan_angle = atan2( m_pts_info_vec[ pt_angle_index ].pt_2d_img( 1 ), m_pts_info_vec[ pt_angle_index ].pt_2d_img( 0 ) ) * 57.3;
                        scan_angle = scan_angle + 180.0;
                    }`

Fellowflyingsky, 2022-04-10:
将拐点之间的点分为内外两类，分别求得点的极坐标

Fellowflyingsky, 2022-04-10:
> in livox_feature_extractor.cpp 497 colum
>
> ` // TODO: handle this case. screen_out << "First point should be normal!!!" << std::endl;
>
> ```
>                 pt_info->pt_2d_img << 0.01, [...]

将拐点之间的点分为内外两类，分别求得点的极坐标