hku-mars/loam_livox #121: catkin_make errors with noetic && Ubuntu 20.04 LTS

Origami404, 2022-09-06:
Solve by change like this:

```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 9e33a03..9cb887d 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -2,7 +2,8 @@ cmake_minimum_required(VERSION 2.8.3)
 project(loam_livox)

 set(CMAKE_BUILD_TYPE "Release")
-set(CMAKE_CXX_FLAGS "-std=c++14")
+#set(CMAKE_CXX_FLAGS "-std=c++14")
+set(CMAKE_CXX_STANDARD 14)
 set(CMAKE_CXX_FLAGS_RELEASE "-O3 -Wall -g")

 find_package(catkin REQUIRED COMPONENTS
diff --git a/source/laser_feature_extractor.hpp b/source/laser_feature_extractor.hpp
index 390491e..ed75aef 100644
--- a/source/laser_feature_extractor.hpp
+++ b/source/laser_feature_extractor.hpp
@@ -39,7 +39,8 @@

 #include <cmath>
 #include <nav_msgs/Odometry.h>
-#include <opencv/cv.h>
+//#include <opencv/cv.h>
+#include <opencv2/opencv.hpp>
 #include <pcl/filters/voxel_grid.h>
 #include <pcl/kdtree/kdtree_flann.h>
 #include <pcl/point_cloud.h>
diff --git a/source/read_camera.cpp b/source/read_camera.cpp
index f99d897..9d92295 100644
--- a/source/read_camera.cpp
+++ b/source/read_camera.cpp
@@ -22,10 +22,10 @@ int main( int argc, char **argv )
       ROS_ERROR("Open Camera error! exit node");
         return -1;
     }
-    cap.set(CV_CAP_PROP_SETTINGS, 1); //opens camera properties dialog
+    cap.set(cv::CAP_PROP_SETTINGS, 1); //opens camera properties dialog

-    cap.set( CV_CAP_PROP_FRAME_WIDTH, 320 );
-    cap.set( CV_CAP_PROP_FRAME_HEIGHT, 240 );
+    cap.set( cv::CAP_PROP_FRAME_WIDTH, 320 );
+    cap.set( cv::CAP_PROP_FRAME_HEIGHT, 240 );
     std::cout << "MJPG: " << cap.set( cv::CAP_PROP_FOURCC, cv::VideoWriter::fourcc( 'M', 'J', 'P', 'G' ) ) << std::endl;
     std::cout << "~~~ Read camera OK ~~~" << std::endl;
     cv::Mat frame;
```

582796566, 2022-11-14:
> Solve by change like this:
>
> ```diff
> diff --git a/CMakeLists.txt b/CMakeLists.txt
> index 9e33a03..9cb887d 100644
> --- a/CMakeLists.txt
> +++ b/CMakeLists.txt
> @@ -2,7 +2,8 @@ [...]

/home/xdh/livox_ws/src/loam_livox/source/ceres_pose_graph_3d.hpp:293:12: error: ‘LocalParameterization’ is not a member of ‘ceres’
  293 |     ceres::LocalParameterization *quaternion_local_parameterization =
      |            ^~~~~~~~~~~~~~~~~~~~~
/home/xdh/livox_ws/src/loam_livox/source/ceres_pose_graph_3d.hpp:293:35: error: ‘quaternion_local_parameterization’ was not declared in this scope
  293 |     ceres::LocalParameterization *quaternion_local_parameterization =
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/xdh/livox_ws/src/loam_livox/source/ceres_pose_graph_3d.hpp:294:20: error: ‘EigenQuaternionParameterization’ in namespace ‘ceres’ does not name a type
  294 |         new ceres::EigenQuaternionParameterization;
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/xdh/livox_ws/src/loam_livox/source/ceres_pose_graph_3d.hpp:317:18: error: ‘class ceres::Problem’ has no member named ‘SetParameterization’; did you mean ‘SetParameterLowerBound’?
  317 |         problem->SetParameterization( pose_begin_iter->second.q.coeffs().data(),
      |                  ^~~~~~~~~~~~~~~~~~~
      |                  SetParameterLowerBound
/home/xdh/livox_ws/src/loam_livox/source/ceres_pose_graph_3d.hpp:319:18: error: ‘class ceres::Problem’ has no member named ‘SetParameterization’; did you mean ‘SetParameterLowerBound’?
  319 |         problem->SetParameterization( pose_end_iter->second.q.coeffs().data(),
      |                  ^~~~~~~~~~~~~~~~~~~
      |                  SetParameterLowerBound
In file included from /home/xdh/livox_ws/src/loam_livox/source/laser_mapping.hpp:68,
                 from /home/xdh/livox_ws/src/loam_livox/source/laser_mapping.cpp:37:
/home/xdh/livox_ws/src/loam_livox/source/point_cloud_registration.hpp: In member function ‘int Point_cloud_registration::find_out_incremental_transfrom(pcl::PointCloud<pcl::PointXYZI>::Ptr, pcl::PointCloud<pcl::PointXYZI>::Ptr, pcl::KdTreeFLANN<pcl::PointXYZI>&, pcl::KdTreeFLANN<pcl::PointXYZI>&, pcl::PointCloud<pcl::PointXYZI>::Ptr, pcl::PointCloud<pcl::PointXYZI>::Ptr)’:
/home/xdh/livox_ws/src/loam_livox/source/point_cloud_registration.hpp:221:20: error: ‘LocalParameterization’ is not a member of ‘ceres’
  221 |             ceres::LocalParameterization *      q_parameterization = new ceres::EigenQuaternionParameterization();
      |                    ^~~~~~~~~~~~~~~~~~~~~
/home/xdh/livox_ws/src/loam_livox/source/point_cloud_registration.hpp:221:49: error: ‘q_parameterization’ was not declared in this scope
  221 |             ceres::LocalParameterization *      q_parameterization = new ceres::EigenQuaternionParameterization();
      |                                                 ^~~~~~~~~~~~~~~~~~
/home/xdh/livox_ws/src/loam_livox/source/point_cloud_registration.hpp:221:74: error: expected type-specifier
  221 |             ceres::LocalParameterization *      q_parameterization = new ceres::EigenQuaternionParameterization();
      |                                                                          ^~~~~
In file included from /home/xdh/livox_ws/src/loam_livox/source/laser_mapping.cpp:37:
/home/xdh/livox_ws/src/loam_livox/source/laser_mapping.hpp: In member function ‘void Laser_mapping::service_loop_detection()’:

hello,after doing this,i meet this new error,can youhelp me?thanks

ZhangPeike, 2024-10-17:
> > > Solve by change like this:
> > > ```diff
> > > diff --git a/CMakeLists.txt b/CMakeLists.txt
> > > index 9e33a03..9cb887d 100644
> > > --- a/CMakeLists.txt
> > > +++ b/CMakeLists.txt
> > > @@ -2,7 +2,8 @@ cmake_minimum_required(VERSION 2.8.3)
> > >  project(loam_livox)
> > > 
> > >  set(CMAKE_BUILD_TYPE "Release")
> > > -set(CMAKE_CXX_FLAGS "-std=c++14")
> > > +#set(CMAKE_CXX_FLAGS "-std=c++14")
> > > +set(CMAKE_CXX_STANDARD 14)
> > >  set(CMAKE_CXX_FLAGS_RELEASE "-O3 -Wall -g")
> > > 
> > >  find_package(catkin REQUIRED COMPONENTS
> > > diff --git a/source/laser_feature_extractor.hpp b/source/laser_feature_extractor.hpp
> > > index 390491e..ed75aef 100644
> > > --- a/source/laser_feature_extractor.hpp
> > > +++ b/source/laser_feature_extractor.hpp
> > > @@ -39,7 +39,8 @@
> > > 
> > >  #include <cmath>
> > >  #include <nav_msgs/Odometry.h>
> > > -#include <opencv/cv.h>
> > > +//#include <opencv/cv.h>
> > > +#include <opencv2/opencv.hpp>
> > >  #include <pcl/filters/voxel_grid.h>
> > >  #include <pcl/kdtree/kdtree_flann.h>
> > >  #include <pcl/point_cloud.h>
> > > diff --git a/source/read_camera.cpp b/source/read_camera.cpp
> > > index f99d897..9d92295 100644
> > > --- a/source/read_camera.cpp
> > > +++ b/source/read_camera.cpp
> > > @@ -22,10 +22,10 @@ int main( int argc, char **argv )
> > >        ROS_ERROR("Open Camera error! exit node");
> > >          return -1;
> > >      }
> > > -    cap.set(CV_CAP_PROP_SETTINGS, 1); //opens camera properties dialog
> > > +    cap.set(cv::CAP_PROP_SETTINGS, 1); //opens camera properties dialog
> > > 
> > > -    cap.set( CV_CAP_PROP_FRAME_WIDTH, 320 );
> > > -    cap.set( CV_CAP_PROP_FRAME_HEIGHT, 240 );
> > > +    cap.set( cv::CAP_PROP_FRAME_WIDTH, 320 );
> > > +    cap.set( cv::CAP_PROP_FRAME_HEIGHT, 240 );
> > >      std::cout << "MJPG: " << cap.set( cv::CAP_PROP_FOURCC, cv::VideoWriter::fourcc( 'M', 'J', 'P', 'G' ) ) << std::endl;
> > >      std::cout << "~~~ Read camera OK ~~~" << std::endl;
> > >      cv::Mat frame;
> > > ```
> > 
> > 
> > /home/xdh/livox_ws/src/loam_livox/source/ceres_pose_graph_3d.hpp:293:12: error: ‘LocalParameterization’ is not a member of ‘ceres’ 293 | ceres::LocalParameterization *quaternion_local_parameterization = | ^~~~~~~~~~~~~~~~~~~~~ /home/xdh/livox_ws/src/loam_livox/source/ceres_pose_graph_3d.hpp:293:35: error: ‘quaternion_local_parameterization’ was not declared in this scope 293 | ceres::LocalParameterization *quaternion_local_parameterization = | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ /home/xdh/livox_ws/src/loam_livox/source/ceres_pose_graph_3d.hpp:294:20: error: ‘EigenQuaternionParameterization’ in namespace ‘ceres’ does not name a type 294 | new ceres::EigenQuaternionParameterization; | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ /home/xdh/livox_ws/src/loam_livox/source/ceres_pose_graph_3d.hpp:317:18: error: ‘class ceres::Problem’ has no member named ‘SetParameterization’; did you mean ‘SetParameterLowerBound’? 317 | problem->SetParameterization( pose_begin_iter->second.q.coeffs().data(), | ^~~~~~~~~~~~~~~~~~~ | SetParameterLowerBound /home/xdh/livox_ws/src/loam_livox/source/ceres_pose_graph_3d.hpp:319:18: error: ‘class ceres::Problem’ has no member named ‘SetParameterization’; did you mean ‘SetParameterLowerBound’? 319 | problem->SetParameterization( pose_end_iter->second.q.coeffs().data(), | ^~~~~~~~~~~~~~~~~~~ | SetParameterLowerBound In file included from /home/xdh/livox_ws/src/loam_livox/source/laser_mapping.hpp:68, from /home/xdh/livox_ws/src/loam_livox/source/laser_mapping.cpp:37: /home/xdh/livox_ws/src/loam_livox/source/point_cloud_registration.hpp: In member function ‘int Point_cloud_registration::find_out_incremental_transfrom(pcl::PointCloudpcl::PointXYZI::Ptr, pcl::PointCloudpcl::PointXYZI::Ptr, pcl::KdTreeFLANNpcl::PointXYZI&, pcl::KdTreeFLANNpcl::PointXYZI&, pcl::PointCloudpcl::PointXYZI::Ptr, pcl::PointCloudpcl::PointXYZI::Ptr)’: /home/xdh/livox_ws/src/loam_livox/source/point_cloud_registration.hpp:221:20: error: ‘LocalParameterization’ is not a member of ‘ceres’ 221 | ceres::LocalParameterization * q_parameterization = new ceres::EigenQuaternionParameterization(); | ^~~~~~~~~~~~~~~~~~~~~ /home/xdh/livox_ws/src/loam_livox/source/point_cloud_registration.hpp:221:49: error: ‘q_parameterization’ was not declared in this scope 221 | ceres::LocalParameterization * q_parameterization = new ceres::EigenQuaternionParameterization(); | ^~~~~~~~~~~~~~~~~~ /home/xdh/livox_ws/src/loam_livox/source/point_cloud_registration.hpp:221:74: error: expected type-specifier 221 | ceres::LocalParameterization * q_parameterization = new ceres::EigenQuaternionParameterization(); | ^~~~~ In file included from /home/xdh/livox_ws/src/loam_livox/source/laser_mapping.cpp:37: /home/xdh/livox_ws/src/loam_livox/source/laser_mapping.hpp: In member function ‘void Laser_mapping::service_loop_detection()’:
> > hello,after doing this,i meet this new error,can youhelp me?thanks
> 
> hi, i ran into the same errors. The problem is that some names changed in ceres. I had to replace some names in the files from the error log, but i cant remember on each name i changed, so start with: ceres::LocalParameterization is now ceres::Manifold. [opencv/opencv_contrib#3218](https://github.com/opencv/opencv_contrib/issues/3218) If you cant solve it, i could look again into those files. Just let me know (i also use noetic & 20.04 with newest pcl, ceres and so on)
> 
> Edit: see also old - ceres::LocalParameterization *q_parameterization = new ceres::EigenQuaternionParameterization();
> 
> new - ceres::LocalParameterization *q_parameterization = new ceres::EigenQuaternionManifold();
> 
> old - problem.SetParameterization(transformVec.at(i).rotationData() new - problem.SetManifold(transformVec.at(i).rotationData(), quaternionParameterization);

I met the same problem being solved by your solution. Could you give the complete key to the problem of ceres. Thanks.

ZhangPeike, 2024-10-17:
> hi, i ran into the same errors. The problem is that some names changed in ceres. I had to replace some names in the files from the error log, but i cant remember on each name i changed, so start [...]

Your solution made it! There're some ceres objects to fix, could you offer the full reference, thanks very much!

Mubashir-alam, 2024-12-01:
@Origami404 which version of OpenCV are you using, I am using `CV_VERSION is 4.10.0` I did not found `opencv/cv.h` I searched for the file in `/usr/local/include/opencv2` which I compiled and install from source code